mcneel/rhino3dm #118: No `constructor defined` for ...

gtalarico, 2019-09-06:
Ran into this error when trying to create `Plane` and `Interval`
My goal was to create a `rc.PlaneSurface`

Are there any other ways of creating these objects in rhino3dm python?

```
>>> import rhino3dm as rc

>>> rc.Plane()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: rhino3dm._rhino3dm.Plane: No constructor defined!

>>> rc.Plane(rc.Point3d(0,0,0), rc.Vector3d(1,0,0))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: rhino3dm._rhino3dm.Plane: No constructor defined!
>>>

mcneel-build, 2019-09-06:
We're experimenting with using Github Issues for our open source projects. Meanwhile, [RH3DM-47](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-47) has also been created on our main issue tracker.

sbaer, 2019-09-06:
No, I need to add constructors for Plane. Shouldn't take too long

mcneel-build, 2019-09-06:
Closed via [RH3DM-47](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-47)

closed 2019-09-06 by mcneel-build