mcneel/rhino3dm #121: Install Issue

Natetronn, 2019-09-09:
Hi,

```
➜  ~ python --version
Python 3.7.3
```

```
➜  ~ pip --version
pip 19.2.3 from /home/{username}/.local/lib/python3.7/site-packages/pip (python 3.7)
```
`Ubunter 18.0.4.3`

Might be me and my setup but, thought I'd report to see if it's something with rhino3dm itself. Looking over the errors I'm not sure how to track down the issues myself.

I get the following error if I run `pip install rhino3dm`, run it in Blender 2.8 python/pip version of python (where I want it) and or the direct release here on github (v0.1.6). Any thoughts on this by chance?

Thanks!

```
➜  ~ pip install rhino3dm
Collecting rhino3dm
  Using cached https://files.pythonhosted.org/packages/2d/4d/933b108e41988ad3987e5e1545b85a639578ec30ab332086dc13170645d3/rhino3dm-0.7.1.tar.gz
Building wheels for collected packages: rhino3dm
  Building wheel for rhino3dm (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-fm04txvn/rhino3dm/setup.py'"'"'; __file__='"'"'/tmp/pip-install-fm04txvn/rhino3dm/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-028ni8kn --python-tag cp37
       cwd: /tmp/pip-install-fm04txvn/rhino3dm/
  Complete output (122 lines):
  /usr/lib/python3.7/distutils/dist.py:274: UserWarning: Unknown distribution option: 'long_description_content_type'
    warnings.warn(msg)
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.7
  creating build/lib.linux-x86_64-3.7/rhino3dm
  copying src/rhino3dm/__init__.py -> build/lib.linux-x86_64-3.7/rhino3dm
  running egg_info
  writing src/rhino3dm.egg-info/PKG-INFO
  writing dependency_links to src/rhino3dm.egg-info/dependency_links.txt
  writing top-level names to src/rhino3dm.egg-info/top_level.txt
  reading manifest file 'src/rhino3dm.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching '*.pyd' under directory 'src/rhino3dm'
  warning: no files found matching '*.so' under directory 'src/rhino3dm'
  writing manifest file 'src/rhino3dm.egg-info/SOURCES.txt'
  copying src/rhino3dm/__init__.pyi -> build/lib.linux-x86_64-3.7/rhino3dm
  copying src/rhino3dm/py.typed -> build/lib.linux-x86_64-3.7/rhino3dm
  running build_ext
  Traceback (most recent call last):
    File "/tmp/pip-install-fm04txvn/rhino3dm/setup.py", line 32, in run
      out = subprocess.check_output(['cmake', '--version'])
    File "/usr/lib/python3.7/subprocess.py", line 395, in check_output
      **kwargs).stdout
    File "/usr/lib/python3.7/subprocess.py", line 472, in run
      with Popen(*popenargs, **kwargs) as process:
    File "/usr/lib/python3.7/subprocess.py", line 775, in __init__
      restore_signals, start_new_session)
    File "/usr/lib/python3.7/subprocess.py", line 1522, in _execute_child
      raise child_exception_type(errno_num, err_msg, err_filename)
  FileNotFoundError: [Errno 2] No such file or directory: 'cmake': 'cmake'
  
  During handling of the above exception, another exception occurred:
  
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-fm04txvn/rhino3dm/setup.py", line 147, in <module>
      include_package_data=True
    File "/home/nate/.local/lib/python3.7/site-packages/setuptools/__init__.py", line 129, in setup
[... 196 lines cut ...]
      File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/home/nate/.local/lib/python3.7/site-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib/python3.7/distutils/command/install.py", line 589, in run
        self.run_command('build')
      File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3.7/distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-fm04txvn/rhino3dm/setup.py", line 36, in run
        ", ".join(e.name for e in self.extensions))
    RuntimeError: CMake must be installed to build the following extensions: rhino3dm/_rhino3dm
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-fm04txvn/rhino3dm/setup.py'"'"'; __file__='"'"'/tmp/pip-install-fm04txvn/rhino3dm/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-v5ca8dv9/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.

```

mcneel-build, 2019-09-09:
We're experimenting with using Github Issues for our open source projects. Meanwhile, [RH3DM-48](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-48) has also been created on our main issue tracker.

jesterKing, 2019-09-09:
@Natetronn the error log states you need to have CMake installed. Apart from that you also need a compiler tool chain to be able to build the code.

Natetronn, 2019-09-09:
Thanks for both your replies, jesterKing!

I should have picked those out from the error and addressed it myself, sorry.

I've gotten a bit further, fixing error code as I go but, have run into this one and I'm not sure how to address it. Could it be my dependency versions aren't lining up properly?

`cmake version 3.15.3`

```
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```
`Python 3.7.3`
`pip 19.2.3 from /home/nate/.local/lib/python3.7/site-packages/pip (python 3.7)`

Any thoughts on this one by chance?
```

➜  rhino3dm git:(master) sudo python3.7 setup.py bdist            
/usr/lib/python3.7/distutils/dist.py:274: UserWarning: Unknown distribution option: 'long_description_content_type'
  warnings.warn(msg)
running bdist
running bdist_dumb
running build
running build_py
running egg_info
writing src/rhino3dm.egg-info/PKG-INFO
writing dependency_links to src/rhino3dm.egg-info/dependency_links.txt
writing top-level names to src/rhino3dm.egg-info/top_level.txt
reading manifest file 'src/rhino3dm.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching '*.pyd' under directory 'src/rhino3dm'
warning: no files found matching '*.so' under directory 'src/rhino3dm'
writing manifest file 'src/rhino3dm.egg-info/SOURCES.txt'
running build_ext
extdir = /home/nate/Desktop/rhino/rhino3dm/build/lib.linux-x86_64-3.7/rhino3dm
sourcedir/home/nate/Desktop/rhino/rhino3dm
Python Compile
-- pybind11 v2.3.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nate/Desktop/rhino/rhino3dm/build/temp.linux-x86_64-3.7
[  0%] Building CXX object CMakeFiles/_rhino3dm.dir/bindings/bindings.cpp.o
In file included from /home/nate/Desktop/rhino/rhino3dm/src/bindings/../lib/opennurbs/opennurbs.h:53:0,
                 from /home/nate/Desktop/rhino/rhino3dm/src/bindings/bindings.h:19,
                 from /home/nate/Desktop/rhino/rhino3dm/src/bindings/bindings.cpp:1:
/home/nate/Desktop/rhino/rhino3dm/src/bindings/../lib/opennurbs/opennurbs_lock.h:118:44: error: use of deleted function 'std::atomic<int>::atomic(const std::atomic<int>&)'
   std::atomic<int> m_lock_value = ON_Lock::UnlockedValue;
                                            ^
In file included from /home/nate/Desktop/rhino/rhino3dm/src/bindings/../lib/opennurbs/opennurbs_system.h:595:0,
                 from /home/nate/Desktop/rhino/rhino3dm/src/bindings/../lib/opennurbs/opennurbs.h:29,
                 from /home/nate/Desktop/rhino/rhino3dm/src/bindings/bindings.h:19,
                 from /home/nate/Desktop/rhino/rhino3dm/src/bindings/bindings.cpp:1:
/home/linuxbrew/.linuxbrew/Cellar/gcc/5.5.0_4/include/c++/5.5.0/atomic:616:7: note: declared here
       atomic(const atomic&) = delete;
       ^
/home/linuxbrew/.linuxbrew/Cellar/gcc/5.5.0_4/include/c++/5.5.0/atomic:620:17: note:   after user-defined conversion: constexpr std::atomic<int>::atomic(std::atomic<int>::__integral_type)
       constexpr atomic(__integral_type __i) noexcept : __base_type(__i) { }
                 ^
CMakeFiles/_rhino3dm.dir/build.make:75: recipe for target 'CMakeFiles/_rhino3dm.dir/bindings/bindings.cpp.o' failed
make[2]: *** [CMakeFiles/_rhino3dm.dir/bindings/bindings.cpp.o] Error 1
CMakeFiles/Makefile2:77: recipe for target 'CMakeFiles/_rhino3dm.dir/all' failed
make[1]: *** [CMakeFiles/_rhino3dm.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Traceback (most recent call last):
  File "setup.py", line 147, in <module>
    include_package_data=True
  File "/home/nate/.local/lib/python3.7/site-packages/setuptools/__init__.py", line 129, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/usr/lib/python3.7/distutils/command/bdist.py", line 143, in run
    self.run_command(cmd_name)
  File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/usr/lib/python3.7/distutils/command/bdist_dumb.py", line 81, in run
    self.run_command('build')
  File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/usr/lib/python3.7/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 45, in run
    self.build_extension(ext)
  File "setup.py", line 96, in build_extension
    system("make")
  File "setup.py", line 20, in system
    raise RuntimeError('The command "{}" exited with {}'.format(cmd, rc))
RuntimeError: The command "make" exited with 2
```

jesterKing, 2019-09-10:
@pearswj didn't this error come up already before?

Anyway, I think you need a newer GCC. I believe @pearswj tested it with GCC 7.4 or newer.

jesterKing, 2019-09-10:
I just double-checked on my Linux Mint 18.3 box - GCC 7.4.0 is used to compile, and works without problems.

pearswj, 2019-09-10:
Yep! `gcc` 7.4 is what's available in the Ubuntu 18.04 repositories, so go with that. You'll need `g++` too.

We have a [CI build](https://github.com/mcneel/rhino3dm/blob/4771b00ae5e7771a12b2fe5be25f3c3fdc35a7cf/.circleci/config.yml#L48) that runs on Ubuntu 18.04 and uses [this command](https://github.com/mcneel/dockerfiles/blob/73c649e2de8270fae895c4ac0bbcf934f275a90b/rhino3dm-dev/Dockerfile#L5) to fetch all the build tools (for python 2/3 and .net builds, so probably overkill for just python)...

``` commandline
$ sudo apt install -y git python3 python3-dev python3-pip python python-dev python-pip uuid-dev cmake clang wget
```

In this case, `python3-pip` (used to install `setuptools`) seems to take care of installing `gcc` and `g++` along with a boat load of other stuff. Maybe we can streamline this a bit...

Looks like `gcc` is pre-installed on Linux Mint. Looks like we need to update the docs!

Natetronn, 2019-09-10:
Thanks so much for the help! Going to give this a go.

Of course, in the middle of it all I thought I'd remove python 3.6, as there seemed to be what I thought was a conflict between it and 3.7, or at least I wasn't sure it wasn't using one when it was suppose to be using the other etc. And then of course I stayed up late into the night reinstalling my machine :-(

**Do not uninstall python 3.6!**

Lesson learned. Sure helped that I had a separate /home on a separate drive and created an initial setup document for things like installing my wifi adapter, changing F keys on an Apple keyboard and general things like that. At least things are freshened up a bit.

Anyway, I'm back up and running.

I also installed python-apt as I was still getting that error:

`sudo apt install  python3-apt`

But it still wasn't finding apt_pkg so, I ran the following:

```
cd /usr/lib/python3/dist-packages
sudo ln -s apt_pkg.cpython-36m-x86_64-linux-gnu.so apt_pkg.so
```
Then to fix the last issue I needed to get more specific with python-dev:

`sudo apt install python3.7-dev`

Then I was on my way. Wohooo!

```
➜  ~ pip install rhino3dm              
Collecting rhino3dm
  Using cached https://files.pythonhosted.org/packages/2d/4d/933b108e41988ad3987e5e1545b85a639578ec30ab332086dc13170645d3/rhino3dm-0.7.1.tar.gz
Building wheels for collected packages: rhino3dm
  Building wheel for rhino3dm (setup.py) ... done
  Created wheel for rhino3dm: filename=rhino3dm-0.7.1-cp37-cp37m-linux_x86_64.whl size=4107579 sha256=2700899f97ee0f89f387c6730ec630abda027a45e3c0e5a14e9283cb43afe221
  Stored in directory: /home/nate/.cache/pip/wheels/6a/d4/d5/1935693025136ce818c2dc44e8b0c44db51a442b772b839fa3
Successfully built rhino3dm
Installing collected packages: rhino3dm
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/usr/local/lib/python3.7/dist-packages/rhino3dm-0.7.1.dist-info'
Consider using the `--user` option or check the permissions.
```

```
➜  ~ pip install rhino3dm --user
Collecting rhino3dm
Installing collected packages: rhino3dm
Successfully installed rhino3dm-0.7.1
```

Thinking I was good to go at this point I started up Blender and tried turning on import_3dm and am now getting the following:

[image]

Natetronn, 2019-09-10:
Okay, I think I may have fixed it. You had a class PolyCurve (uppercase C) in both /rhino3dm/build/lib.linux-x86_64-3.7/rhino3dm/__init__.pyi (and in 2.7) but, it was looking for Polycurve (lowercase c) in curve.py.

jesterKing, 2019-09-11:
@Natetronn the PolyCurve problem is know. I haven't had time to push the simple fix for it (rhino3dm  0.7.0 introduced the small braking change).

Since you're able to install now I'm going to close this issue. Thank you for reporting the motions you went through!

closed 2019-09-11 by jesterKing

mcneel-build, 2019-09-11:
Closed via [RH3DM-48](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-48)