mcneel/rhino3dm #123: TypeScript type declarations

danvk, 2019-09-30:
I'm using the rhino3dm.js library in a TypeScript project. It would be amazing if McNeel could generate TypeScript type declarations for this library, either bundled with the package or on [DefinitelyTyped][2].

I've written some simple declarations for the `File3dm` class (included below), but something official would be much better, especially because the API docs are [not always clear][1] about the parameter/return types for methods.

```ts
// rhino3dm.d.ts
declare namespace RhinoAPI {
  // See https://mcneel.github.io/rhino3dm/javascript/api/File3dm.html
  class File3dm {
    applicationDetails: string;
    applicationName: string;
    applicationUrl: string;
    createdBy: string;
    lastEditedBy: string;
    revision: number;
    startSectionComments: string;

    dimstyles(): File3dmDimStyleTable;
    instanceDefinitions(): unknown;
    layers(): File3dmLayerTable;
    materials(): unknown;
    objects(): File3dmObjectTable;
    settings(): File3dmSettings;

    static fromByteArray(array: Uint8Array): File3dm;
  }

  // https://developer.rhino3d.com/api/RhinoCommon/html/T_Rhino_FileIO_CommonComponentTable_1.htm
  interface CommonComponentTable<T> {
    count(): number;
    clear(): void;
    delete(item: T): void;
    findId(id: Guid): T;
  }

  interface File3dmCommonComponentTable<T> extends CommonComponentTable<T> {
    add(item: T): void;
  }

  interface File3dmLayerTable extends File3dmCommonComponentTable<Layer> {
    findName(name: string, parentId: Guid): Layer | null;
    findIndex(i: number): Layer;
  }

  interface File3dmDimStyleTable extends File3dmCommonComponentTable<DimensionStyle> {
    findIndex(index: number): DimensionStyle;
    get(): DimensionStyle;
  }

  interface File3dmObjectTable {
    count: number;
    get(i: number): File3dmObject;
  }

  class Layer {
    id: Guid;
    name: string;
    parentLayerId: Guid;
  }

  class File3dmObject {
    attributes(): ObjectAttributes;
    geometry(): unknown;
  }

  interface ObjectAttributes {
    layerIndex: number;
  }

  class DimensionStyle {
    // (lots of properties here)
  }

  interface File3dmSettings {
    modelAbsoluteTolerance: number;
    modelAngleToleranceDegrees: number;
    modelAngleToleranceRadians: number;
    modelBasePoint: [number, number, number];
    modelRelativeTolerance: number;
    modelUnitSystem: number;
    modelUrl: string;
    pageAbsoluteTolerance: number;
    pageAngleToleranceDegrees: number;
    pageAngleToleranceRadians: number;
    pageRelativeTolerance: number;
    pageUnitSystem: number;
    renderSettings(): unknown;
  }

  type Guid = string;

  interface API {
    File3dm: typeof File3dm;
  }
}

declare module 'rhino3dm' {
  function loadRhino3dm(): RhinoAPI.API;
  export = loadRhino3dm;
}

```

[1]: https://mcneel.github.io/rhino3dm/javascript/api/File3dm.html#instanceDefinitions
[2]: https://github.com/DefinitelyTyped/DefinitelyTyped

mcneel-build, 2019-09-30:
Linked with [RH3DM-53](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-53)

sbaer, 2019-09-30:
We do generate typescript declarations. Are you getting rhino3dm.js from npm?

Here is a discussion of the topic
https://discourse.mcneel.com/t/rhino3dm-js-typescript-declaration-file/87468

closed 2019-09-30 by sbaer

mcneel-build, 2019-09-30:
Closed via [RH3DM-53](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-53)

danvk, 2019-09-30:
Looks like I was just on an old version. Sorry! And thanks for the types :)

nicolas-chaulet, 2021-03-08:
Hey! I am using Rhino3dm in a typescript ap and I am having trouble with the fromByteArray method, indeed, it looks like it is not correctly typed, the input is missing.
https://mcneel.github.io/rhino3dm/javascript/api/File3dm.html#.fromByteArray