mcneel/rhino3dm #126: InstanceDefinition missing Attributes/Methods

lfertig, 2019-10-06:
Instance definitions don't expose several attributes and methods. The 3 most important ones to provide basic functionality would probably be InstanceDefinition.Id, InstanceDefinition.Name and .GetObjectIds(), as listed in the .net docs: [https://developer.rhino3d.com/api/RhinoCommon/html/T_Rhino_Geometry_InstanceDefinitionGeometry.htm](url).

forum thread:
[https://discourse.mcneel.com/t/rhino3dm-instancedefinitions/89904/5?u=fertig](url)

mcneel-build, 2019-10-06:
Linked with [RH3DM-55](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-55)

lfertig, 2019-10-09:
First things first, I am new to contributing something here and assume people are fare more experienced with most matters. Hence I'm open to any guidance, (constructive) criticism and if someone feels this should rather be handled by someone else, please feel free to say so.

I had a moment to look at the python binding and would propose to store the ObjectIds in an std::vector<BND_UUID> which then can be bound as an opaque stl container by pybind. This way the resulting object in python behaves like a simple list of uuids:

[image]

As for the conversion from ON_SimpeArray<ON_UUID>to std::vector<BND_UUID>, do i see this correctly that this HAS to be done by first converting it to an array of strings (suuid) and from there to the respective python/wasm types? 

I attached the two files of my minimal example:
[main.zip](https://github.com/mcneel/rhino3dm/files/3709232/main.zip)

closed 2019-10-23 by sbaer

mcneel-build, 2019-12-19:
Closed via [RH3DM-55](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-55)

PR #133 (2019-10-23): implements BND_InstanceDefinitionGeometry.GetObjectIds(), exposes .Name and .Id
Fixes #126 / RH3DM-55

It is currently not possible to read the Name, Id or the child objects of an InstanceDefinition from the .3dm file. The PR exposes the necessary variables and adds two functions to do so.

*exposes BND_InstanceDefinitionGeometry.Name and .Id

*implements BND_InstanceDefinitionGeometry.GetObjectIds() to return a std::vector<BND_UUID> containing the IDs of all child objects, that can be passed on by pybind

*to do so, the PR adds a converter to bnd_uuid.h/cpp that converts from ON_SimpleArray<O [... cut]