mcneel/rhino3dm #132: Blender 2.81 - Getting ModuleNotFoundError: No module named 'rhino3dm' on MacOS after doing special Mac install instructions

kalmdown, 2019-10-22:
Did the special MacOS pip instructions. 
```
osxkmochmbp15:~ kmochel$ cd /Applications/Blender.app/Contents/Resources/2.81/python
osxkmochmbp15:python kmochel$ ls ./bin
pip3		pip3.7		python3.7m
```
Still receiving this error when trying to activate in Blender prefs. 
[image]

Blender - Version 2.81 (2.81 2019-10-16)
MacOS - 10.14.6 (18G95)

mcneel-build, 2019-10-22:
Linked with [RH3DM-57](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-57)

jesterKing, 2019-10-22:
This is really an issue that should be reported at https://github.com/jesterKing/import_3dm :)

mcneel-build, 2019-10-22:
Closed via [RH3DM-57](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-57)

closed 2019-10-22 by mcneel-build

kalmdown, 2019-10-22:
Done! - K

----------
Karl Mochel
Interaction, Interface, Information, Experience Design
linkedin.com/in/kalmdown
<https://www.linkedin.com/in/kalmdown?lipi=urn%3Ali%3Apage%3Ad_flagship3_profile_view_base_contact_details%3BpyZvd%2BHjSLSo3xELyIygSw%3D%3D>