mcneel/rhino3dm #156: Support for React

philipblaquiere, 2019-11-14:
Attempting to initialize rhino3dm.js in a React App results in WASM compile fail. 

(On Chrome 78.0.3904.97, React 16.11.0, Rhino3dm 0.8.1)

> CompileError: WebAssembly.instantiate(): expected magic word 00 61 73 6d, found 3c 21 44 4f @+0

Is there a way around this?

mcneel-build, 2019-11-14:
Linked with [RH3DM-63](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-63)

karimi, 2019-11-14:
Hi @philipblaquiere 
Have a look at this:
https://discourse.mcneel.com/t/in-browser-python-interpreter/92055/7?u=mkarimi

philipblaquiere, 2019-12-17:
This is more a workaround rather than a solution. 

The biggest pain is to require an async context to load the library.

sbaer, 2020-02-17:
wasm requires async loading. That is not specific to react and is a browser requirement

closed 2020-02-17 by sbaer