mcneel/rhino3dm #166: JavaScript: Adding a layer and geometry to a doc results in invalid .3dm file

pearswj, 2019-12-11:
In JavaScript I'm unable to create a document, add a layer and add an object (geometry) to that layer. The resulting file throws a "... is not a Rhino m_archive" error when opened in Rhino.

The following change to the [sketch2d sample](https://github.com/mcneel/rhino3dm/blob/bde1bf6028fd7a6c3d681cf2c3885810ac1d8df0/docs/javascript/samples/sketch2d/app.js) reproduces the error.

```diff
--- a/docs/javascript/samples/sketch2d/app.js
+++ b/docs/javascript/samples/sketch2d/app.js
@@ -35,6 +35,13 @@ function download3dm() {
   }
 
   let doc = new _rhino3dm.File3dm();
+
+  // create a custom layer
+  let layer = new _rhino3dm.Layer();
+  layer.name = 'example';
+  layer.color = { r: 248, g: 131, b: 121, a: 255 };
+  doc.layers().add(layer);
+
   for(let i=0; i<_model.curves.length;i++) {
     doc.objects().add(_model.curves[i], null);
   }
```

This python script works just fine...

```python
import rhino3dm
import os

model = rhino3dm.File3dm()

layer = rhino3dm.Layer()
layer.Color = (255, 0, 0, 127)
layer.Name = 'example'

model.Layers.Add(layer)

model.Objects.AddPoint(0, 0, 0)

desktop = '/Users/will/Desktop'
filename = 'py_layer_test.3dm'
path = os.path.join(desktop, filename)

model.Write(path, 6)

print(path)
```

mcneel-build, 2019-12-11:
Linked with [RH3DM-68](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-68)

closed 2019-12-13 by sbaer

mcneel-build, 2019-12-16:
Closed via [RH3DM-68](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-68)

PR #167 (2019-12-13): fix layer addition routine
Fixes RH3DM-68
Closes #166