mcneel/rhino3dm #171: Mesh method 'toThreejsJSON' converts mesh to Y-Axes up

philipblaquiere, 2019-12-16:
When converting a rhino mesh to a threejs mesh the function assumes the target axes (Threejs) has the Y axes as a vertical, which is the default for Threejs.

There should be a way to configure the target axes configuration when converting from a Rhino mesh to a Threejs mesh, either as a function parameter or a new toThreejsJSON function.

mcneel-build, 2019-12-16:
Linked with [RH3DM-69](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-69)

sbaer, 2019-12-17:
This is possible, but I'm curious as to why you would want this

philipblaquiere, 2019-12-17:
If we want to use Rhino as a source of truth, we need to replicate its default axes definition in Threejs. This becomes especially evident when ghx definitions utilize inputs (like x,y,z positions or points) that were captured through interactions with elements in Threejs (eg. plane). It prevents unecessary conversions from system to system rather axes definition to axes definition.

That said it is also possible to modify the default 'up' in Threejs, making it easy to replicate rhino's axes.

pearswj, 2019-12-17:
> That said it is also possible to modify the default 'up' in Threejs, making it easy to replicate rhino's axes.

@philipblaquiere I tried to implement this but couldn’t get it to work. Do you have a working example?

philipblaquiere, 2019-12-18:
> I tried to implement this but couldn’t get it to work. Do you have a working example?

@pearswj Have you tried the following before any threejs component is initialized?

`THREE.Object3D.DefaultUp = new THREE.Vector3(0,0,1);`

pearswj, 2019-12-19:
@philipblaquiere Thanks. I tried adding this to the [basic viewer sample](https://github.com/mcneel/rhino3dm/blob/master/docs/javascript/samples/viewer/01_basic/index.html) but it didn't appear to have any effect. Would you mind giving it a go?

sbaer, 2019-12-19:
I have this working @pearswj. Should have a PR very soon

mcneel-build, 2019-12-19:
Closed via [RH3DM-69](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-69)

closed 2019-12-19 by mcneel-build

sbaer, 2019-12-19:
https://mcneel.github.io/rhino3dm/javascript/samples/viewer/01_basic/
has been updated to use the `DefaultUp` call and `thThreejsJSON` does not transform the output

https://mcneel.github.io/rhino3dm/javascript/samples/viewer/02_advanced/
does not use `DefaultUp` and calls an overload of `toThreejsJSON` that accepts a bool to determine if rotation to Y-up should occur

pearswj, 2019-12-20:
Ah, I wasn't seeing the effect of setting `DefaultUp` with the old `camera.position.z = 40`, but now I am. Thank you both for teaching me something new!