mcneel/rhino3dm #176: bootstrap script supports javascript build

dbelcher, 2019-12-20:
As far of supporting Scripts-to-rule-them-all (https://github.com/github/scripts-to-rule-them-all), start with bootstrap scripts for javascript builds.

`$ bootstrap.py --platform js`

should run on:

- macOS
- Linux (Ubuntu)
- Windows (if we can swing it)

and also support all the `--check` switches necessary to check all those tools for each OS.

mcneel-build, 2019-12-20:
Linked with [RH3DM-70](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-70)

dbelcher, 2020-01-14:
This should work now for javascript

mcneel-build, 2020-01-28:
Closed via [RH3DM-70](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-70)

dbelcher, 2020-01-29:
Looks to me like I have more work to do on Windows Linux Subsystem as well.

closed 2020-04-20 by mcneel-build