mcneel/rhino3dm #190: CircleCI: use scripts for javascript build

dbelcher, 2020-01-17:
As of https://github.com/mcneel/rhino3dm/pull/189, we can now use:
`script/boostrap.py --platform js` (optional)
followed by:
`script/setup.py --platform js --overwrite`
followed by:
`script/build.py --platform js --overwrite --verbose`

This is (🤞) a more robust version of of _src/build_javascript.py_.

Once this is done, I believe _src/build_javascript.py_ can be deleted.

Documentation is updated here:
https://github.com/mcneel/rhino3dm/blob/master/docs/javascript/RHINO3DM-BUILD.JS.md

I have only tested these scripts on macOS.  If there are problems on linux, please let me know.

mcneel-build, 2020-01-17:
Linked with [RH3DM-73](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-73)

fraguada, 2021-05-20:
UPDATE 2021.05.20: Circle CI still uses `src/build_javascript.py `while the build instructions use `bootstrap, setup, build.py`

pearswj, 2021-05-20:
I didn't do a good job of documenting why I didn't yet adopt the build scripts for the CI build, but #278 sounds familiar. Anyway, we ought to move to GitHub Actions to make it easier for other in the team to help maintain the builds (#373).

PR #396 (2021-06-01): CI Workflow - Builds JS with scripts
Fixes #190
commit 61c432c9 (2021-06-01): CI Workflow - Builds JS with scripts (#396)

Fixes #190
closed 2021-06-01 by fraguada