mcneel/rhino3dm #194: Emscripten 1.39.6 breaks JavaScript build

pearswj, 2020-01-22:
At the end of the build I got the following error:

```
...
[100%] Linking CXX executable rhino3dm.html
shared:ERROR: Customizing EXPORT_NAME requires that the HTML be customized to use that name (see https://github.com/emscripten-core/emscripten/issues/10086)
CMakeFiles/rhino3dm.dir/build.make:6517: recipe for target 'rhino3dm.html' failed
make[2]: *** [rhino3dm.html] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/rhino3dm.dir/all' failed
make[1]: *** [CMakeFiles/rhino3dm.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

This looks to be related to emscripten-core/emscripten@3043df918fa5a478431ae77a3fbd28841ac5661c. It happens using both the old and new build scripts.

Tested on Ubuntu 18.04

```commandline
$ python bootstrap.py -p js

Checking JavaScript Dependencies...

Checking Git...
  This system is running Git 2.17.1
  We are currently using Git 2.17.1
 OK: Git version 2.17.1 found.

Checking Python...
  This system is running Python 2.7.15
  We are currently using Python 2.7.17
 WARNING: Python version 2.7.15 found, an older version. We are currently using 2.7.17.  On Ubuntu, you can install python 2 using sudo apt install python

Checking Emscripten...
  This system is running Emscripten 1.39.5
  We are currently using Emscripten 1.39.5
 OK: Emscripten version 1.39.5 found.

Checking CMake...
  This system is running CMake 3.10.2
  We are currently using CMake 3.16.2
 WARNING: CMake version 3.10.2 found, an older version. We are currently using 3.16.2.  On Ubuntu, you can install CMake using: sudo snap install cmake --classic
```

mcneel-build, 2020-01-22:
Linked with [RH3DM-74](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-74)

pearswj, 2020-11-10:
Just a note to self for when looking into this... try 1.39.16 or later (emscripten-core/emscripten#10697, [thanks Chuck](https://discourse.mcneel.com/t/await-rhino3dm-js-wasm-import/101585/3?u=will)).

Or [v2](https://github.com/emscripten-core/emscripten/pull/11856)...

sbaer, 2020-12-17:
The same error occurs with v2.0
We don't need html output from the compile so it looks like tweaking the CMakeLists.txt to output .wasm.js instead will fix things. I was able to get a compile with 2.0 after this change. Testing changes now with the current emscripten that we are targeting.

mcneel-build, 2020-12-17:
Closed via [RH3DM-74](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-74)

closed 2020-12-17 by mcneel-build