mcneel/rhino3dm #209: Rhino3DM JIT compilation dependent?

SiriusA7, 2020-02-01:
Are any elements of Rhino3DM relying on “just-in-time”(JIT) compilation at runtime?
Same question framed differently: does Rhino3DM support “Ahead-of-time”(AOT) compilation?

mcneel-build, 2020-02-01:
Linked with [RH3DM-75](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-75)

mcneel-build, 2020-02-01:
Closed via [RH3DM-75](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-75)

closed 2020-02-01 by mcneel-build

sbaer, 2020-02-01:
We have compiled Rhino3dmIO for iOS which requires AOT compilation. Rhino3dm is just an updated version and we should soon have iOS compiles available

SiriusA7, 2020-02-01:
@sbaer thank you for the response!
Is the same possible for android?
I'm concerned for android builds from unity3d because they force il2cpp, which is AOT, as a scripting back-end for builds.

sbaer, 2020-02-01:
This is something we'll need to check with Unity about as it is very specific to their architecture. I don't know the answer

SiriusA7, 2020-02-02:
I've been dealing with this issue for the past few weeks now.
I finally got building unity il2cpp to work with Rhino3dmIO.Android, but at runtime the only relevant error stream is this:

```
ArgumentException: Invalid path
  at System.IO.Path.GetDirectoryName (System.String path) [0x00000] in <00000000000000000000000000000000>:0 
  at UnsafeNativeMethods.Init () [0x00000] in <00000000000000000000000000000000>:0 
  at UnsafeNativeMethods..cctor () [0x00000] in <00000000000000000000000000000000>:0 
  at Rhino.Runtime.InteropWrappers.SimpleArrayPoint3d..ctor () [0x00000] in <00000000000000000000000000000000>:0 
  at Rhino.Geometry.Polyline.CreateInscribedPolygon (Rhino.Geometry.Circle circle, System.Int32 sideCount) [0x00000] in <00000000000000000000000000000000>:0 
  at TestRhino.GenerateGeo () [0x00000] in <00000000000000000000000000000000>:0 
  at TestRhino.Update () [0x00000] in <00000000000000000000000000000000>:0 
 Rethrow as TypeInitializationException: The type initializer for 'UnsafeNativeMethods' threw an exception.
  at Rhino.Runtime.InteropWrappers.SimpleArrayPoint3d..ctor () [0x00000] in <00000000000000000000000000000000>:0 
  at Rhino.Geometry.Polyline.CreateInscribedPolygon (Rhino.Geo
```

I can send a sample project if it helps

sbaer, 2020-02-04:
Attaching a sample project to this github issue may be best. Maybe someone with more knowledge of il2cpp can help with this.