mcneel/rhino3dm #210: Accessing Custom User Defined Attributes

Dorgam, 2020-02-09:
Hi, I am trying to access the customly defined attributes here:

[image]

How can I access these Key/Value pairs using rhino3dm.js?

Thanks

mcneel-build, 2020-02-09:
Linked with [RH3DM-76](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-76)

sbaer, 2020-02-11:
This can be accessed with getUserString and setUserString on object attributes from a File3dm.
```
let file = File3dm.fromByteArray(...)
let objects = file.objects()
for (let i=0; i<objects.count; i++) {
  let obj = objects.get(i)
  let attributes = obj.attributes()
  console.log(attributes.getUserString('Floor2Floor'))
  attributes.delete
  obj.delete()
}
objects.delete()
file.delete()
```

closed 2020-02-11 by sbaer

mcneel-build, 2020-02-11:
Closed via [RH3DM-76](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-76)