mcneel/rhino3dm #212: unable to build on linux

maxnbk, 2020-02-14:
same error with both py2.7 and py3.7, gcc-4, gcc-6, llvm-9
```
Failed to build rhino3dm
Installing collected packages: rhino3dm
    Running setup.py install for rhino3dm ... error
    ERROR: Command errored out with exit status 1:
     command: /sw/packages/external/python/2.7.13/platform-linux/arch-x86_64/os-CentOS-7/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-BYVT7S/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-BYVT7S/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-X_psU6/install-record.txt --single-version-externally-managed --prefix /sw/dev/maxnbk/packages/rhino3dm/0.10.0/platform-linux/arch-x86_64/os-CentOS-7/python-2.7 --no-compile
         cwd: /tmp/pip-req-build-BYVT7S/
    Complete output (108 lines):
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-2.7
    creating build/lib.linux-x86_64-2.7/rhino3dm
    copying src/rhino3dm/__init__.py -> build/lib.linux-x86_64-2.7/rhino3dm
    running egg_info
    writing src/rhino3dm.egg-info/PKG-INFO
    writing top-level names to src/rhino3dm.egg-info/top_level.txt
    writing dependency_links to src/rhino3dm.egg-info/dependency_links.txt
    reading manifest file 'src/rhino3dm.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no files found matching '*.pyd' under directory 'src/rhino3dm'
    warning: no files found matching '*.so' under directory 'src/rhino3dm'
    writing manifest file 'src/rhino3dm.egg-info/SOURCES.txt'
    copying src/rhino3dm/__init__.pyi -> build/lib.linux-x86_64-2.7/rhino3dm
    copying src/rhino3dm/py.typed -> build/lib.linux-x86_64-2.7/rhino3dm
    warning: build_py: byte-compiling is disabled, skipping.

    running build_ext
    extdir = /tmp/pip-req-build-BYVT7S/build/lib.linux-x86_64-2.7/rhino3dm
    sourcedir/tmp/pip-req-build-BYVT7S
    -- The C compiler identification is GNU 6.3.1
    -- The CXX compiler identification is GNU 6.3.1
    -- Check for working C compiler: /opt/rh/devtoolset-6/root/usr/bin/cc
    -- Check for working C compiler: /opt/rh/devtoolset-6/root/usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Check for working CXX compiler: /opt/rh/devtoolset-6/root/usr/bin/c++
    -- Check for working CXX compiler: /opt/rh/devtoolset-6/root/usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    Python Compile
    -- Found PythonInterp: /sw/packages/external/python/2.7.13/platform-linux/arch-x86_64/os-CentOS-7/bin/python (found version "2.7.13")
    -- Found PythonLibs: /sw/packages/external/python/2.7.13/platform-linux/arch-x86_64/os-CentOS-7/lib/libpython2.7.so
    -- Performing Test HAS_CPP14_FLAG
    -- Performing Test HAS_CPP14_FLAG - Success
    -- pybind11 v2.4.3
    -- Performing Test HAS_FLTO
    -- Performing Test HAS_FLTO - Success
    -- LTO enabled
    -- Configuring done
    -- Generating done
    -- Build files have been written to: /tmp/pip-req-build-BYVT7S/build/temp.linux-x86_64-2.7
    make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
    Scanning dependencies of target _rhino3dm
    [  0%] Building CXX object CMakeFiles/_rhino3dm.dir/bindings/base64.cpp.o
    [  0%] Building CXX object CMakeFiles/_rhino3dm.dir/bindings/bindings.cpp.o
    In file included from /tmp/pip-req-build-BYVT7S/src/bindings/../lib/opennurbs/opennurbs.h:53:0,
                     from /tmp/pip-req-build-BYVT7S/src/bindings/bindings.h:19,
                     from /tmp/pip-req-build-BYVT7S/src/bindings/bindings.cpp:1:
    /tmp/pip-req-build-BYVT7S/src/bindings/../lib/opennurbs/opennurbs_lock.h:118:44: error: use of deleted function ‘std::atomic<int>::atomic(const std::atomic<int>&)’
       std::atomic<int> m_lock_value = ON_Lock::UnlockedValue;
                                                ^~~~~~~~~~~~~
    In file included from /tmp/pip-req-build-BYVT7S/src/bindings/../lib/opennurbs/opennurbs_system.h:595:0,
                     from /tmp/pip-req-build-BYVT7S/src/bindings/../lib/opennurbs/opennurbs.h:29,
                     from /tmp/pip-req-build-BYVT7S/src/bindings/bindings.h:19,
                     from /tmp/pip-req-build-BYVT7S/src/bindings/bindings.cpp:1:
    /opt/rh/devtoolset-6/root/usr/include/c++/6.3.1/atomic:616:7: note: declared here
           atomic(const atomic&) = delete;
           ^~~~~~
    /opt/rh/devtoolset-6/root/usr/include/c++/6.3.1/atomic:620:17: note:   after user-defined conversion: constexpr std::atomic<int>::atomic(std::atomic<int>::__integral_type)
           constexpr atomic(__integral_type __i) noexcept : __base_type(__i) { }
                     ^~~~~~
    CMakeFiles/_rhino3dm.dir/build.make:75: recipe for target 'CMakeFiles/_rhino3dm.dir/bindings/bindings.cpp.o' failed
    make[5]: *** [CMakeFiles/_rhino3dm.dir/bindings/bindings.cpp.o] Error 1
    CMakeFiles/Makefile2:77: recipe for target 'CMakeFiles/_rhino3dm.dir/all' failed
    make[4]: *** [CMakeFiles/_rhino3dm.dir/all] Error 2
    Makefile:83: recipe for target 'all' failed
    make[3]: *** [all] Error 2
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-BYVT7S/setup.py", line 152, in <module>
        include_package_data=True
      File "/sw/packages/external/setuptools/41.4.0/python-2.7/lib/python2.7/site-packages/setuptools/__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "/sw/packages/external/python/2.7.13/platform-linux/arch-x86_64/os-CentOS-7/lib/python2.7/distutils/core.py", line 151, in setup
        dist.run_commands()
      File "/sw/packages/external/python/2.7.13/platform-linux/arch-x86_64/os-CentOS-7/lib/python2.7/distutils/dist.py", line 953, in run_commands
        self.run_command(cmd)
      File "/sw/packages/external/python/2.7.13/platform-linux/arch-x86_64/os-CentOS-7/lib/python2.7/distutils/dist.py", line 972, in run_command
        cmd_obj.run()
      File "/sw/packages/external/setuptools/41.4.0/python-2.7/lib/python2.7/site-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/sw/packages/external/python/2.7.13/platform-linux/arch-x86_64/os-CentOS-7/lib/python2.7/distutils/command/install.py", line 563, in run
        self.run_command('build')
      File "/sw/packages/external/python/2.7.13/platform-linux/arch-x86_64/os-CentOS-7/lib/python2.7/distutils/cmd.py", line 326, in run_command
        self.distribution.run_command(command)
      File "/sw/packages/external/python/2.7.13/platform-linux/arch-x86_64/os-CentOS-7/lib/python2.7/distutils/dist.py", line 972, in run_command
        cmd_obj.run()
      File "/sw/packages/external/python/2.7.13/platform-linux/arch-x86_64/os-CentOS-7/lib/python2.7/distutils/command/build.py", line 127, in run
        self.run_command(cmd_name)
      File "/sw/packages/external/python/2.7.13/platform-linux/arch-x86_64/os-CentOS-7/lib/python2.7/distutils/cmd.py", line 326, in run_command
        self.distribution.run_command(command)
      File "/sw/packages/external/python/2.7.13/platform-linux/arch-x86_64/os-CentOS-7/lib/python2.7/distutils/dist.py", line 972, in run_command
        cmd_obj.run()
      File "/tmp/pip-req-build-BYVT7S/setup.py", line 45, in run
        self.build_extension(ext)
      File "/tmp/pip-req-build-BYVT7S/setup.py", line 99, in build_extension
        system("make")
      File "/tmp/pip-req-build-BYVT7S/setup.py", line 20, in system
        raise RuntimeError('The command "{}" exited with {}'.format(cmd, rc))
    RuntimeError: The command "make" exited with 2
```

mcneel-build, 2020-02-14:
Linked with [RH3DM-77](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-77)

pearswj, 2020-02-15:
Try updating GCC to version 7 or 8. Our Python 2 CI builds currently run on Ubuntu with GCC 7.4 and our Python 3 builds run on Debian with GCC 8.3.

bareya, 2020-02-18:
Change of compiler has little to do with error that is happening here:
```
    /tmp/pip-req-build-BYVT7S/src/bindings/../lib/opennurbs/opennurbs_lock.h:118:44: error: use of deleted function ‘std::atomic<int>::atomic(const std::atomic<int>&)’
       std::atomic<int> m_lock_value = ON_Lock::UnlockedValue;

pearswj, 2020-02-28:
Thanks for the suggestion @bareya – I'll pass it along to my colleagues who look after opennurbs.

In the meantime, I managed to install the rhino3dm package via pip (Python 3) on Centos 7 [using gcc 7.3.1](https://www.softwarecollections.org/en/scls/rhscl/devtoolset-7/).

```commandline
yum install centos-release-scl
yum install devtoolset-7-gcc*
scl enable devtoolset-7 bash
pip3 install -vv rhino3dm
exit
```

closed 2020-02-28 by pearswj

mcneel-build, 2020-02-28:
Closed via [RH3DM-77](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-77)