mcneel/rhino3dm #228: I'm I crazy??

ruifortes, 2020-03-07:
Why the hell can't I wrap this??

```
var rhino3dm = require('rhino3dm')

function init() {
	return new Promise((resolve, reject) => {
		rhino3dm()
		.then(rhino => {
			// resolve("some other return") this would resolve the promise
			resolve(rhino) // but not this
		})
	})
}

init()
.then(rhino => {
	// never gets here?!
	let sphere = new rhino.Sphere([1,2,3,], 12)
	console.log(sphere.radius)
})
.catch(err => {
	//...thenable returned by rhino doesn't define catch
})
```

Where is this weirdness coming from?
What am I missing here?

Thanks

muka, 2020-04-04:
This works for the `then` but there is no `catch` and I cannot find a way to detect issues on initialization.

```
    return new Promise((resolve, reject) => {
      try { rhino3d().then(rhino => resolve({ rhino })) } catch (e) { reject(e) }
    }).then(({ rhino }) => {
      console.log("OK rhino", rhino);
    }).catch((e) => {
      console.error("rhino3dm failure: %s", e)
    });
```

pearswj, 2020-12-27:
This was caused by some weirdness in the rhino3dm.js produced by Emscripten. I just [published v0.14.0 to npm](https://www.npmjs.com/package/rhino3dm) which was built using the latest version of Emscripten (#339) – `rhino3dm()` should now return a proper _thenable_ that can also be awaited.

Thanks @cdriesler for [the tip](https://discourse.mcneel.com/t/await-rhino3dm-js-wasm-import/101585/3?u=will)!

closed 2020-12-27 by pearswj