mcneel/rhino3dm #241: How to generate new nurbs curve when move the control point of the start and end control point

friends110110, 2020-03-28:
In CAD software , I interpolate a nurbs curve , now I want to move the control point of the start and end point , it means move the control point of the start and end control point ? Then how to generate a new nurbs curfve after move the start and end point of the nurbs curve, could return interpolate points of the new nurbs curve.

mcneel-build, 2020-03-28:
Linked with [RH3DM-85](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-85)

friends110110, 2020-03-28:
@mcneel-build could you provide some code(c++ ,java, python code) to move the control point of the start and end point, and evaluate the nurbs curve?

sbaer, 2020-03-30:
What programming language are you trying to work in? Are you using rhino3dm or is this a general question about Rhino?

friends110110, 2020-04-01:
@sbaer mainly java， but other programming language is also ok. Could you provide some examples?

sbaer, 2020-04-13:
I just realized that my 2d curve builder sample may be able to help you out here.

A new curve is created from a set of points at this location in the code
https://github.com/mcneel/rhino3dm/blob/master/docs/javascript/samples/sketch2d/app.js#L106

Here is a live version of the sample
https://mcneel.github.io/rhino3dm/javascript/samples/sketch2d/

closed 2020-04-13 by sbaer