mcneel/rhino3dm #270: bootstrap.py --platform js throws error on Emscripten check on WLS

dbelcher, 2020-04-24:
`python bootstrap.py --platform js` throws an error when checking for Emscripten on Windows Linux Subsystem (but NOT on Ubuntu proper).

Traceback (from both Python 2 and 3):
```
Traceback (most recent call last):
   File "bootstrap.py", line 1049, in <module>
      main()
   File "bootstrap.py", line 1023, in main
      check_handler(target_platform, build_tools)
   File "bootstrap.py", line 787, in check_handler
      check_emscripten(build_tools["emscripten"])
   File "bootstrap.py", line 384, in check_emscripten
      print_error_message(err)
   File "bootstrap.py", line 80, in print_error_message
      print(bcolors.BOLD + bcolors.FAIL + error_prefix.upper() + bcolors.ENDC + bcolors.FAIL + error_message +
TypeError: must be str, not bytes
```

mcneel-build, 2020-04-24:
Linked with [RH3DM-92](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-92)

mcneel-build, 2020-04-24:
Closed via [RH3DM-92](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-92)

closed 2020-04-24 by mcneel-build