mcneel/rhino3dm #275: Library for javascript dependent on browser

kretz, 2020-05-12:
The rhino3dm module for javascript uses the `btoa` function on this line;
https://github.com/mcneel/rhino3dm/blob/fcb6ee63e2924310fc5c4fb3894b1014d00aa66b/docs/javascript/samples/resources/compute.rhino3d.js#L3658

That function is declared on the Window object of the browser and will not work on e.g. node.js.

For those who neea a workaround for node.js could be to declare the function if it does not exist:

```
if (typeof btoa === 'undefined') {
  global.btoa = function (str) {
    return Buffer.from(str, 'binary').toString('base64');
  };
}
```

mcneel-build, 2020-05-12:
Linked with [RH3DM-97](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-97)

pearswj, 2020-05-12:
Hey! When I added node.js support to `grasshopper.evaluateDefinition()` I was using `fs` to load the .gh and .ghx files during testing.

```js
let buffer = fs.readFileSync('example.ghx')
let definition = new Uint8Array(buffer)
```

By not specifying the encoding when reading the file, the same code works for both .gh and .ghx files.

If you have other reasons to pass the .ghx definition as a string then please open a pull request and we'll get this sorted! The js client is generated by a .net project in the compute.rhino3d repo. Here's the right place to add it...

https://github.com/mcneel/compute.rhino3d/blob/864067c87caacac292e767a18338f2bd2e274df0/src/compute.client/JavascriptClient.cs#L221

kretz, 2020-05-12:
Ok, thanks. I actually do it as you suggest already, but I thought this was a bug. 

This example here sort of indicated it could be a possibility that should work?
https://github.com/mcneel/rhino3dm/blob/master/docs/javascript/samples/compute/RESTHopper/Extrusions/app.js

I also saw this (https://discourse.mcneel.com/t/failed-to-execute-btoa-on-window/97009) today, so decided to create an issue. 

But if sending it as an array is the preferred way for the API then I'm happy and we can discard this issue. It was just not clear to me. :)

commit 5ff665e5 (2020-05-15): Use uint8array in grasshopper samples

Both .gh and .ghx files can be read as an array buffer and fed to evaluateDefinition() as a Uint8Array. No need to show a alternative method for passing the definition as a string (which can cause issues if the .ghx file has a BOM).

Closes #275
closed 2020-05-15 by pearswj