mcneel/rhino3dm #276: bootstrap.py --check emscripten throws error on WSL Ubuntu 18.04

dbelcher, 2020-05-12:
To Reproduce

1. Launch WSL Ubuntu 18.04.
1. Run `python script/bootstrap.py --check emscripten`

The following error occurs
```
TypeError: must be str, not bytes
```

Be careful fixing this as I believe it can break Ubuntu 18.04 proper :/

mcneel-build, 2020-05-12:
Linked with [RH3DM-98](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-98)

pearswj, 2020-05-12:
Working fine here. I think it's best to "installing" emscripten inside the WSL, rather than using an existing clone of emscripten-core/emsdk elsewhere on the Windows system. When sharing one clone between Windows and WSL I found I had to do a thorough git clean and start again each time I switched to get the emscripten toolchain working correctly on the current platform.

[image]

sbaer, 2020-05-12:
Would it be best to just add emscripten as a submodule?

fraguada, 2021-05-31:
@dbelcher @pearswj is this obsolete?

pearswj, 2021-05-31:
Yep, I don't think we need to push WSL now that we have scripts that work with emscripten installed on Windows (#370).

closed 2021-05-31 by pearswj