mcneel/rhino3dm #278: build.py --platform js success check may be faulty

dbelcher, 2020-05-12:
I recently failed to build the javascript library using:
```
python scripts/build.py --platform js -o -v
```
because there was a build error.  However, the script reported success (and the error was suppressed).  I suspect the check for success is looking for a file and it is finding an older build and reporting success when it should not.  This can be tested by artificially introducing a build error in the code.

mcneel-build, 2020-05-12:
Linked with [RH3DM-100](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-100)

fraguada, 2023-10-05:
Recent testing confirms that the appropriate files are removed with the -o flag

fraguada, 2024-12-02:
I've noticed that this is an issue on dotnet builds. build.py declares success, but there are errors reported.

fraguada, 2025-02-28:
Fixed the js error checking by ensuring the rhino3dm.js and rhino3dm.wasm are deleted.

closed 2025-02-28 by fraguada