mcneel/rhino3dm #279: Use CMake 3.10.2 on Ubuntu and WSL

dbelcher, 2020-05-12:
Only use the latest that is easy to install with sudo apt-get.  We don't need a later version yet, as far as I know.  This should simplify installation instructions.

mcneel-build, 2020-05-12:
Linked with [RH3DM-101](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-101)

pearswj, 2020-05-12:
I think I was wrong in claiming that the version listed in the CMakeLists.txt files is the minimum. For a while now we've listed 3.12.2 as the minimum in the docs.

https://github.com/mcneel/rhino3dm/blob/f7b5c8cb31d9b66bbc362ea4e4a5a0c7fb3383db/docs/javascript/RHINO3DM-BUILD.JS.md#L16

Would be awesome if it worked though!

sbaer, 2020-05-12:
We need at least the version of CMake that creates VS2019 projects.

pearswj, 2020-05-13:
I think that means [3.14.0](https://gitlab.kitware.com/cmake/cmake/-/merge_requests/2789) is the minimum.

pearswj, 2020-12-26:
Ubuntu 20.04 LTS provides CMake 16.6.3 via `apt` so I think we're now fine with the current suggested version in "Current Build Tools" :+1:

closed 2020-12-26 by pearswj