mcneel/rhino3dm #285: Publish a nuget package

pearswj, 2020-05-22:
We should replace the Rhino3dmIO packages with a new Rhino3dm package built from this repo. The "desktop" package should support both .NET Framework and Core.

mcneel-build, 2020-05-22:
Linked with [RH3DM-102](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-102)

pearswj, 2020-05-25:
Provisional nuget package published based on 79f7726.

https://www.nuget.org/packages/Rhino3dm/0.1.0

StuYarrow, 2020-10-27:
Is it planned to include the win-x86 native lib in the NuGet package?  This is a show stopper for me, my tests are passing on Win64 and Linux (which is awesome!), but I need Win32 as well.

pearswj, 2020-12-27:
@StuYarrow I just published a v0.3.0 NuGet package which includes a Win32 native lib.

pearswj, 2020-12-27:
Added nuspecs for iOS and Android in #346. Needs testing and #348 addressing before publishing.

fraguada, 2021-08-11:
@pearswj I think we can close this and keep working on Android in #348.

closed 2021-08-11 by fraguada