mcneel/rhino3dm #299: Use with Python 3.8.3

KeithSloan, 2020-07-13:
I am trying to use rhino3dm with FreeCAD. The version of python that FreeCAD uses is 3.8.3

If I try import rhino3dm it does not seem to recognise the module

`>>> import rhino3dm
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Applications/FreeCAD_0.19.app/Contents/Resources/lib/python3.8/rhino3dm/__init__.py", line 5, in <module>
    from ._rhino3dm import *
ModuleNotFoundError: No module named 'rhino3dm._rhino3dm'
>>> print(sys.version)
3.8.3 | packaged by conda-forge | (default, Jun  1 2020, 17:26:49) 
[Clang 9.0.1 ]
>>>

mcneel-build, 2020-07-13:
Linked with [RH3DM-107](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-107)

sbaer, 2020-07-16:
I do already provide pre-built binaries supporting 3.8 on Windows, but not on Mac yet. Have you tried building this project under 3.8 for Mac?

KeithSloan, 2020-07-16:
Was hoping to avoid having to build from source. Would also mean advising users of my workbench that they would have to build rhino3dm from source if on a Mac, not sure many could.

sbaer, 2020-07-16:
Was just checking to see if this works. I plan on eventually shipping 3.8 prebuilt binaries for Mac at some point in time, but probably not this week

KeithSloan, 2020-07-16:
If I can find some time, I will try, but very busy at present.

KeithSloan, 2020-07-29:
Okay I built from source and using python3 from the command line it all works.

Now I am trying to use under FreeCAD and it required the module to be installed in a certain directory.

Having run python3 setup.py bdist what do I do to also get a copy in a different directory?

Thanks

sbaer, 2020-08-03:
What directory does freecad want rhino3dm installed in?

KeithSloan, 2020-08-03:
It can go in any of the directories that FreeCAD knows about.

One can get a list of these from the FreeCAD python console with`
>>> import sys
>>> print(sys.path)
['/Users/keithsloan/Library/Preferences/FreeCAD/Mod/GDML', '/Users/keithsloan/Library/Preferences/FreeCAD/Mod/dxf_library', '/Users/keithsloan/Library/Preferences/FreeCAD/Mod/ImportNURBS', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Raytracing', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Material', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Import', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Part', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Tux', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/PartDesign', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Measure', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Points', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/AddonManager', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Arch', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Idf', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Inspection', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Image', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Robot', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Surface', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/OpenSCAD', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Start', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Sketcher', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Show', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/MeshPart', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Path', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/TechDraw', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Web', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Complete', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Spreadsheet', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Fem', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Test', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Drawing', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/ReverseEngineering', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Draft', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod/Mesh', '/Applications/FreeCAD_0.19.app/Contents/Resources/Mod', '/Applications/FreeCAD_0.19.app/Contents/Resources/lib', '/Applications/FreeCAD_0.19.app/Contents/Resources/Ext', '/Applications/FreeCAD_0.19.app/Contents/Resources/bin', '/Applications/FreeCAD_0.19.app/Contents/Resources/lib/python38.zip', '/Applications/FreeCAD_0.19.app/Contents/Resources/lib/python3.8', '/Applications/FreeCAD_0.19.app/Contents/Resources/lib/python3.8/lib-dynload', '/Applications/FreeCAD_0.19.app/Contents/Resources/lib/python3.8/site-packages', '/Users/keithsloan/Library/Preferences/FreeCAD/Macro', '/Applications/FreeCAD_0.19.app/Contents/Resources/Macro', '/Users/keithsloan/Library/Application Support/FreeCAD/Mod']
>>> 

I would opt for /Applications/FreeCAD_0.19.app/Contents/Resources/lib/python3.8/site-packages'

This is on. a Mac. Other operating systems may well give a different list, so it would be good to be able to
specify the directory.

sbaer, 2020-08-03:
Does Mac actually let you place executable binaries inside the app bundle these days?

You should be able to pip install to a specific directory

https://stackoverflow.com/questions/17216689/pip-install-python-package-into-a-specific-directory-other-than-the-default-inst/23435273

KeithSloan, 2020-08-03:
But I had to build from source as needed versions not available via pip

sbaer, 2020-08-03:
You can pip install a local file

KeithSloan, 2020-08-04:
Not sure how to do that with a local file from source

tried

pip3 install rhino3dm /Applications/FreeCAD_0.19.app/Contents/Resources/lib/python3.8/site-packages

Got

ERROR: Directory '/Applications/FreeCAD_0.19.app/Contents/Resources/lib/python3.8/site-packages' is not installable. Neither 'setup.py' nor 'pyproject.toml' found.

KeithSloan, 2020-08-04:
I found the tar.gz archive and ran

pip3 install ./dist/rhino3dm-0.12.0.macosx-10.15-x86_64.tar.gz -t /Applications/FreeCAD_0.19.app/Contents/Resources/lib/python3.8/site-packages
Processing ./dist/rhino3dm-0.12.0.macosx-10.15-x86_64.tar.gz
    ERROR: Command errored out with exit status 1:
     command: /usr/local/opt/python@3.8/bin/python3.8 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/cj/z259fmwd41dgzl8mq8nppwd40000gn/T/pip-req-build-bi70mnxz/setup.py'"'"'; __file__='"'"'/private/var/folders/cj/z259fmwd41dgzl8mq8nppwd40000gn/T/pip-req-build-bi70mnxz/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/cj/z259fmwd41dgzl8mq8nppwd40000gn/T/pip-pip-egg-info-gsgzrpii
cwd: /private/var/folders/cj/z259fmwd41dgzl8mq8nppwd40000gn/T/pip-req-build-bi70mnxz/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/lib/python3.8/tokenize.py", line 392, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/private/var/folders/cj/z259fmwd41dgzl8mq8nppwd40000gn/T/pip-req-build-bi70mnxz/setup.py'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

sbaer, 2020-08-04:
This may be Mac blocking installation into an app bundle, but I'm not sure. I just installed Python 3.8 on my Mac and will try to get a compile up on pypi.org soon

sbaer, 2020-08-08:
I just published a Mac 3.8 binary to PyPi.org

closed 2020-08-08 by sbaer