mcneel/rhino3dm #302: Bindings: ViewInfo.viewport

fraguada, 2020-08-08:
I can get the properties of a `ViewInfo` object, but cannot seem to get the `.viewport` in js:
[image]

mcneel-build, 2020-08-08:
Linked with [RH3DM-109](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-109)

fraguada, 2020-08-08:
Seems it is commented out here: https://github.com/mcneel/rhino3dm/blob/master/src/bindings/bnd_3dm_settings.cpp#L145

fraguada, 2021-06-07:
It seems the count property is used twice: https://github.com/mcneel/rhino3dm/blob/main/src/bindings/bnd_extensions.cpp#L1543-L1550

```cpp
  class_<BND_File3dmStringTable>("File3dmStringTable")
    .function("count", &BND_File3dmStringTable::Count)
    .function("get", &BND_File3dmStringTable::GetKeyValue)
    .function("getvalue", &BND_File3dmStringTable::GetValueFromKey)
    .function("set", &BND_File3dmStringTable::SetString)
    .function("count", &BND_File3dmStringTable::DocumentUserTextCount)
    .function("delete", &BND_File3dmStringTable::Delete)
    ;
```

I'll try to rebuild without the latter to see if it makes any difference.

jjsolly, 2021-08-06:
Accidentally created a new issue about this same item. I would also like to access .viewport information. 

Looking in the Rhino3dm.js API it seems that the ViewportInfo class does exist but as per the above I cannot find out how to access this information from a specific ViewInfo object. I'm trying to access information via namedViews from the 3dm file.

fraguada, 2023-10-10:
this has been included in rhino3dm8.0.0-beta3

closed 2023-10-10 by fraguada

dmaslan, 2023-12-06:
I'm still not able to access this in rhino3dm8.0.0-beta3...am i missing something?
[image]