mcneel/rhino3dm #303: File3dm.strings always empty

fraguada, 2020-08-08:
In a doc with a few test strings, `doc.strings().count()` always returns 0.

mcneel-build, 2020-08-08:
Linked with [RH3DM-110](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-110)

fraguada, 2021-06-07:
Here is a 3dm with two strings:

- `{ key: Foo, value: Bar }`
- `{ key: Hello, value: World }`

[DocumentStrings.zip](https://github.com/mcneel/rhino3dm/files/6609486/DocumentStrings.zip)

fraguada, 2021-06-08:
It seems the count property is used twice: https://github.com/mcneel/rhino3dm/blob/main/src/bindings/bnd_extensions.cpp#L1543-L1550
```
  class_<BND_File3dmStringTable>("File3dmStringTable")
    .function("count", &BND_File3dmStringTable::Count)
    .function("get", &BND_File3dmStringTable::GetKeyValue)
    .function("getvalue", &BND_File3dmStringTable::GetValueFromKey)
    .function("set", &BND_File3dmStringTable::SetString)
    .function("count", &BND_File3dmStringTable::DocumentUserTextCount)
    .function("delete", &BND_File3dmStringTable::Delete)
    ;
```

closed 2021-06-08 by fraguada

kovacsv, 2021-07-16:
Thanks for the fix, @fraguada. Do you now when will be a release available with this fix?

fraguada, 2021-08-04:
@kovacsv I've just published rhino3dm.js 7.7.0 https://www.npmjs.com/package/rhino3dm

kovacsv, 2021-08-05:
Thanks, @fraguada, it works really well!

PR #401 (2021-06-08): Bindings: Fix doc.strings().count() property for js
Fixes #303. 

The property name `count` for doc.strings().count() was used twice, once for doc.strings().count() and then used again for doc.strings().documentUserTextCount(). Fixed by adding `documentUserTextCount` as the property name.