mcneel/rhino3dm #307: library versions

fraguada, 2020-08-28:
(I'm not sure we have any library version information in the compiled versions of the js and py libraries.)

Add a property to the library object which contains the library version. 

Question raised here: https://discourse.mcneel.com/t/3dmloader-for-three-js/107702/3

mcneel-build, 2020-08-28:
Linked with [RH3DM-114](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-114)

pearswj, 2020-08-28:
For python, the library version is defined in [__init__.py](https://github.com/mcneel/rhino3dm/blob/2fc48fe3c3957283c2cd9e7997a790bf91a1e1b7/src/rhino3dm/__init__.py)
```py
import rhino3dm
print(rhino3dm.__version__)
```
I don't think it's defined anywhere in the library for javascript.

fraguada, 2025-02-27:
Added in https://github.com/mcneel/rhino3dm/pull/680

closed 2025-02-27 by fraguada