mcneel/rhino3dm #308: Issue compiling on Ubuntu 20.04LTS

sanchitgulati, 2020-09-03:
Started with installing NET Core SDK on Ubuntu
[Microsoft Guide](https://docs.microsoft.com/en-us/dotnet/core/install/linux-ubuntu)

`wget https://packages.microsoft.com/config/ubuntu/20.04/packages-microsoft-prod.deb -O packages-microsoft-prod.deb`
`sudo dpkg -i packages-microsoft-prod.deb`
`sudo apt-get update;`
 `sudo apt-get install -y apt-transport-https`
 `sudo apt-get update`
 `sudo apt-get install -y dotnet-sdk-2.2`

dotnet 2.2 is no longer supported by Microsoft on ubuntu

I can successfully do
 `sudo apt-get install -y dotnet-sdk-3.1`
 `sudo apt-get install -y dotnet-sdk-2.1`

Since the project is configured to 2.2, I am unable to build.
Looking into Rhino3dm on Linux to setup AWS lambda.

Any plans to upgrade the build system or tips? Thanks

mcneel-build, 2020-09-03:
Linked with [RH3DM-115](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-115)

pearswj, 2020-09-03:
Hey @sanchitgulati, have you tried the [rhino3dm nuget package](https://www.nuget.org/packages/rhino3dm)? It was published with AWS Lambda in mind. I'll leave this issue open to remind me to revisit the build setup and make sure it works on the latest Ubuntu LTS...

pearswj, 2020-12-26:
Project was updated to use .NET Core 3.1 in 3faaed9af0da6d30b26f74f4263f239d5ba0089d (but I forgot to update this...)

closed 2020-12-26 by pearswj