mcneel/rhino3dm #318: How to add normals/texture coordinates?

cedemax, 2020-09-15:
I'm using rhino3dm to support the 3dm file format as an export alternative in my app. 

I can't figure out how to add normals and texture coordinates to a Mesh (which is created from scratch).
Currently I do this:
```
var m = new Rhino.Geometry.Mesh();
m.Vertices.AddVertices(verts);
m.Faces.AddFaces(faces);
```

which gives me a nice mesh, but no normals/textures.

The Mesh object has **TextureCoordinates** and **Normals**, but I can't figure out how they are supposed to be used, since they dont have any way to specify what vertex/triangle they are referring to? I also don't see a way to assign a material to the mesh?

mcneel-build, 2020-09-15:
Linked with [RH3DM-116](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-116)

sbaer, 2020-09-15:
Let's start with normals.

There should be one normal for every vertex in a mesh. If you have normals that you want to apply, then
`m.Normals.Add` or `m.Normals.AddRange` is what you will want to use
If you want to have the normals computed based on vertex and face information, then call `m.Normals.ComputeNormals();`

mcneel-build, 2020-09-15:
Closed via [RH3DM-116](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-116)

closed 2020-09-15 by sbaer

cedemax, 2020-09-15:
Allright, thank you for the quick clarification!

How does one then do different textures coordinates on the shared vertex of adjacent triangles? Or is this not possible? Would one need to have separate vertices for the two triangles?

sbaer, 2020-09-15:
You would need separate vertices if you want separate normals or tcs

cedemax, 2020-09-15:
Awesome, thanks!

Is there some example of how to add/use a texture to the mesh as well?

sbaer, 2020-09-15:
This involves materials. Please ask how-tos on our discourse forum at https://discourse.mcneel.com/ as I'm not the best person to answer this question.