mcneel/rhino3dm #319: mesh=face.GetMesh(rhino3dm.MeshType.Any) return None

yehongyu1234, 2020-10-05:
hi
I try to program the file
[image]
python/samples/getmeshes.py
then return None
[image]
why ?
thanku

mcneel-build, 2020-10-05:
Linked with [RH3DM-118](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-118)

jesterKing, 2020-10-05:
You need to have render (or other) meshes saved with your file: Set your viewport to Rendered mode, save.

Further, have you tried getting meshes for the entire object in one go? Or do you really want to extract the meshes face by face?

yehongyu1234, 2020-10-05:
> You need to have render (or other) meshes saved with your file: Set your viewport to Rendered mode, save. `rhino3dm` does not contain a tesselator.
> 
> Further, have you tried getting meshes for the entire object in one go? Or do you really want to extract the meshes face by face?
@jesterKing 

thanku for replay,
python seems not support directly convert brep to mesh,so I have no method to try getting mesh except through compute_rhino3d.
yes,I want to extract the meshes face by face;
I have tried the sample code file 
docs/python/samples/gemeshes.py  for testing.

`import rhino3dm `
`model = rhino3dm.File3dm.Read('sample2.3dm')`
`brep = model.Objects[0].Geometry`
`face = brep.Faces[0]`
`mesh = face.GetMesh(rhino3dm.MeshType.Any)`
`print (len(mesh.Faces))`

also get "None",that means brep faces can't convert to Mesh Faces with method "GetMesh()" .
or "GetMesh()" is wip?

yehongyu1234, 2020-10-05:
@jesterKing 

Version problem,I try to saving rhino6 as rhino5.It works.
but I think the problem also need to solving

mcneel-build, 2020-10-05:
Closed via [RH3DM-118](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-118)

closed 2020-10-05 by mcneel-build

jesterKing, 2020-10-06:
@yehongyu1234 as I mentioned the `rhino3dm` library does not contain a tesselator. If a file does not contain render meshes for an object then you can't retrieve them either. Using Rhino Compute or Rhino Inside is your only option in getting surfaces meshed.