mcneel/rhino3dm #320: pip install on Windows fails

s3ththompson, 2020-10-06:
```
> pip install rhino3dm
```

Fails on Windows with the following CMake error:

```
CMake Error at lib/draco/CMakeLists.txt:28 (include):
      include could not find load file:

        C:/Users/Seth Thompson/AppData/Local/Temp/pip-install-aj5_5pey/rhino3dm/src/lib/draco/cmake/compiler_flags.cmake
```

It looks like there is no such `draco/cmake` directory in `rhino3dm-0.13.0-tar.gz`.

Installing from the git repo instead worked `pip install --no-clean git+https://github.com/mcneel/rhino3dm.git`.

<hr>

```
> pip --version
pip 20.2.3 from c:\users\seth thompson\appdata\local\programs\python\python39\lib\site-packages\pip (python 3.9)
```

```
> cmake --version
cmake version 3.18.3
```

```
> python --version
Python 3.9.0
```

mcneel-build, 2020-10-06:
Linked with [RH3DM-119](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-119)

pearswj, 2020-12-16:
Fixed in `master` and we'll publish a new version of rhino3dm for python soon (including a wheel for python 3.9 on Windows).

commit d56ce829 (2020-12-16): Add missing draco files to sdist

Fixes #320
closed 2020-12-16 by pearswj