mcneel/rhino3dm #322: ReferenceError: WebAssembly is not defined

PlumpHip, 2020-10-22:
Welcome to Node.js v12.18.0.
Type ".help" for more information.
> rhino3dm = require('rhino3dm')()
no native wasm support detected
.... rhino3dm-0.13.0/lib/node_modules/rhino3dm/rhino3dm.js:9
var Module=typeof rhino3dm!=="undefined"?rhino3dm:{};var moduleOverrides={};var key;for(key in Module){if(Module.hasOwnProperty(key))
....
;while(Module["preInit"].length>0){Module["preInit"].pop()()}}noExitRuntime=true;run();

ReferenceError: WebAssembly is not defined
    at ..../rhino3dm-0.13.0/lib/node_modules/rhino3dm/rhino3dm.js:9:7225

error... ?
crash node and go to shell prompt 

Thanks.

mcneel-build, 2020-10-22:
Linked with [RH3DM-121](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-121)

pearswj, 2020-10-23:
Can you please provide a bit more information? What OS and version are you using? How did you install node.js?

I'm unable to reproduce this error on macOS 10.15.6, with node.js v12.18.0 installed via nvm.

PlumpMath, 2020-10-23:
I'm (its me)
Use nix on macos.
Hmm; 
.
Strangely, in lumo-cljs (clojurescript) it loads fine without a repl crash.  Due to the nature of nix installation.  I think it may be a problem caused by symlinking the location of node_modules... So, I am thinking of trying again in node by installing locally.

PlumpMath, 2020-11-22:
I'm not sure, but...
in node REPL (this problem)
The reason why all the modules were not called.
Maybe. I found out that my node is a gralvm node.
(Unfortunately, I didn't even know what I had developed...) Hahahaha)
I hope this will be a clue to this simple problem-solving.

pearswj, 2020-12-26:
We don't have any experience with GraalVM... I did some quick reading and this looks like a good place to start: https://www.graalvm.org/reference-manual/wasm/. I don't know if it supports loading a WebAssembly from node.js yet, though. We use [Emscripten](https://github.com/emscripten-core/emscripten) to compile rhino3dm to WebAssembly – maybe they can help!

closed 2020-12-26 by pearswj