mcneel/rhino3dm #338: File3dm.toByteArray().buffer full of 0s

fraguada, 2020-12-17:
To make File3dm compatible with the [Transferable API](https://developer.mozilla.org/en-US/docs/Web/API/Transferable) (which is needed to pass docs to workers, as in done in the 3dmLoader), we need to pass an `ArrayBuffer`. These are typically available after creating a `UInt8Array` via `UInt8Array.buffer`. 

`File3dm.toByteArray()` returns a `UInt8Array` ([src](https://github.com/mcneel/rhino3dm/blob/master/src/bindings/bnd_extensions.cpp#L1120-L1142)), but it's `.buffer` data is all 0s.

One workaround I've found is to recreate the `UInt8Array` like so: 

```
let nCrv = rhino.NurbsCurve.create( false, 3, crvPoints)
let doc = new rhino.File3dm()
doc.objects().addCurve(nCrv, null)
let ba = doc.toByteArray(); //ba.buffer full of 0s
let newArr = new Uint8Array(ba); newArr.buffer contains data
```

TODO: Investigate if we can make the `.toByteArray()` method to return a `UIInt8Array `that has the right data in its buffer.

mcneel-build, 2020-12-17:
Linked with [RH3DM-132](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-132)

fraguada, 2020-12-18:
Seems the ArrayBuffer is pointing to all of the wasm memory:

[image]

From other discussions on the WebAssembly discord and emscripten repo, the best approach is just to copy the relevant data by creating a new Uint8Array to get the appropriately sized ArrayBuffer.

closed 2020-12-18 by fraguada