mcneel/rhino3dm #348: Rhino3dm.dll (Android) is much larger than the old Rhino3dmIO.dll

pearswj, 2020-12-23:
In particular, the native libs (armeabi-v7a, arm64-v8a, x86_64, x86) are larger than those produced by the old Rhino3dmIO (gyp) build.

mcneel-build, 2020-12-23:
Linked with [RH3DM-134](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-134)

pearswj, 2020-12-23:
Old...
```
% du -hsc src4/bin/rhino3dmio/build/Release-android/libs/*/*

7.6M	src4/bin/rhino3dmio/build/Release-android/libs/arm64-v8a/librhino3dmio_native.so
5.1M	src4/bin/rhino3dmio/build/Release-android/libs/armeabi-v7a/librhino3dmio_native.so
8.4M	src4/bin/rhino3dmio/build/Release-android/libs/x86/librhino3dmio_native.so
8.1M	src4/bin/rhino3dmio/build/Release-android/libs/x86_64/librhino3dmio_native.so
 29M	total
```

New...
```
% du -hsc src/build/android/libs/*/*

 37M	src/build/android/libs/arm64-v8a/librhino3dm_native.so
 42M	src/build/android/libs/armeabi-v7a/librhino3dm_native.so
 45M	src/build/android/libs/x86/librhino3dm_native.so
 36M	src/build/android/libs/x86_64/librhino3dm_native.so
160M	total
```