mcneel/rhino3dm #350: Support importing rhino3dm.js as an ES6 module

pearswj, 2021-01-20:
We should ship an additional JS wrapper that's compiled using `-s EXPORT_ES6=1` that can be used with the ES6 `import` syntax. This would keep things consistent when using other libraries like three.js.

`import rhino3dm from './rhino3dm.module.js'`

https://github.com/emscripten-core/emscripten/issues/8506 (idea for shipping multiple wrappers)

mcneel-build, 2021-01-20:
Linked with [RH3DM-135](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-135)

pearswj, 2021-03-22:
We've been experimenting with this in the 0.15-beta release which includes both the normal rhino3dm.js and rhino3d.module.js (this is just the rhino3dm.js that's produced when `EXPORT_ES6=1`). Seems to work great!

fraguada, 2021-06-07:
Since 0.16.1 / 7.6.0 we support shipping rhino3dm.js with rhino3dm.module.js.

closed 2021-06-07 by fraguada