mcneel/rhino3dm #353: toByteArray produces invalid 3dm file in some cases

pearswj, 2021-02-09:
Here's a simple node.js script that reproduces the issue...

```js
const rhino3dm = require('rhino3dm')
const compute = require('compute-rhino3d')
const fs = require('fs')

compute.url = 'http://localhost:8081/'

const name = 'box_test.gh'

const buffer = fs.readFileSync(name)
const definition = new Uint8Array(buffer)

async function doStuff() {
    const rhino = await rhino3dm()
    let res = await compute.Grasshopper.evaluateDefinition(definition, [])

    const json = JSON.parse(res.values[0].InnerTree['{ 0; }'][0].data)
    const mesh = rhino.CommonObject.decode(json)

    const doc = new rhino.File3dm()
    doc.objects().add(mesh, null)

    const sphere = new rhino.Sphere([0,0,0], 12)
    
    // add this sphere and the 3dm file will be valid!
    doc.objects().addSphere(sphere, null)

    console.log('saved')
    fs.writeFileSync('box_test.3dm', doc.toByteArray())
}

doStuff()
```

The script relies on a Grasshopper definition to create a single Box (and mesh it).

[box_test.zip](https://github.com/mcneel/rhino3dm/files/5951852/box_test.zip)

mcneel-build, 2021-02-09:
Linked with [RH3DM-138](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-138)

fraguada, 2021-02-09:
Additional info: 
Using the rhino3dm functions to convert the mesh to threejs works fine:
```javascript
let loader = new THREE.BufferGeometryLoader();
let geometry = loader.parse(rhinoObject.toThreejsJSON());
let mesh = new THREE.Mesh(geometry, new THREE.MeshNormalMaterial());
scene.add(mesh);
```

pearswj, 2021-02-10:
@sbaer I was unable to reproduce the bug in python (script below). I was able to write a valid file with only the geometry I got back from solving box_test.gh. I was also able to write a valid empty file by instantiating a `File3dm` and immediately writing it - this isn't possible in JavaScript/node.js.

```python
#!/usr/bin/env python

import compute_rhino3d.Util
import compute_rhino3d.Grasshopper as gh
import rhino3dm
import json
import sys

compute_rhino3d.Util.url = 'http://localhost:5000/'

res = gh.EvaluateDefinition('box_test.gh', [])

data = json.loads(res['values'][0]['InnerTree']['{ 0; }'][0]['data'])

mesh = rhino3dm.CommonObject.Decode(data)

print('Got a mesh with {} vertices and {} faces'.format(
    len(mesh.Vertices), len(mesh.Faces)))

doc = rhino3dm.File3dm()
doc.Objects.Add(mesh)

print('Writing doc with {} object(s)'.format(len(doc.Objects)))

filename = 'box_test.3dm'
if doc.Write(filename, 7):
    print('Saved to: ' + filename)
```

fraguada, 2021-02-17:
@pearswj @sbaer here is another case of this issue popping up: https://discourse.mcneel.com/t/rhino3dm-threejs-geometry-import-failing/118635/6

closed 2021-05-19 by fraguada