mcneel/rhino3dm #357: Can't install in Ubuntu

seghier, 2021-03-24:
Hello, i try to install it in ubuntu without success

```
pip3 install --user rhino3dm
Collecting rhino3dm
  Using cached rhino3dm-0.14.0.tar.gz (5.4 MB)
Building wheels for collected packages: rhino3dm
  Building wheel for rhino3dm (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-asbyc7hq/rhino3dm/setup.py'"'"'; __file__='"'"'/tmp/pip-install-asbyc7hq/rhino3dm/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-2nk7fxjz
       cwd: /tmp/pip-install-asbyc7hq/rhino3dm/
  Complete output (95 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.8
  creating build/lib.linux-x86_64-3.8/rhino3dm
  copying src/rhino3dm/__init__.py -> build/lib.linux-x86_64-3.8/rhino3dm
  running egg_info
  writing src/rhino3dm.egg-info/PKG-INFO
  writing dependency_links to src/rhino3dm.egg-info/dependency_links.txt
  writing top-level names to src/rhino3dm.egg-info/top_level.txt
  reading manifest file 'src/rhino3dm.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching '*.pyd' under directory 'src/rhino3dm'
  warning: no files found matching '*.so' under directory 'src/rhino3dm'
  writing manifest file 'src/rhino3dm.egg-info/SOURCES.txt'
  copying src/rhino3dm/__init__.pyi -> build/lib.linux-x86_64-3.8/rhino3dm
  copying src/rhino3dm/py.typed -> build/lib.linux-x86_64-3.8/rhino3dm
  running build_ext
  extdir = /tmp/pip-install-asbyc7hq/rhino3dm/build/lib.linux-x86_64-3.8/rhino3dm
  sourcedir/tmp/pip-install-asbyc7hq/rhino3dm
  -- The C compiler identification is GNU 5.5.0
  -- The CXX compiler identification is GNU 5.5.0
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: /home/linuxbrew/.linuxbrew/bin/cc - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: /home/linuxbrew/.linuxbrew/bin/c++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  Python Compile
  -- pybind11 v2.6.1
  -- Found PythonInterp: /usr/bin/python3 (found version "3.8.5")
  -- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so
  -- Performing Test HAS_FLTO
  -- Performing Test HAS_FLTO - Success
  -- Configuring done
  -- Generating done
  -- Build files have been written to: /tmp/pip-install-asbyc7hq/rhino3dm/build/temp.linux-x86_64-3.8
  Scanning dependencies of target _rhino3dm
  [  0%] Building CXX object CMakeFiles/_rhino3dm.dir/bindings/base64.cpp.o
  [  0%] Building CXX object CMakeFiles/_rhino3dm.dir/bindings/bindings.cpp.o
  In file included from /tmp/pip-install-asbyc7hq/rhino3dm/src/bindings/../lib/opennurbs/opennurbs.h:53:0,
                   from /tmp/pip-install-asbyc7hq/rhino3dm/src/bindings/bindings.h:19,
                   from /tmp/pip-install-asbyc7hq/rhino3dm/src/bindings/bindings.cpp:1:
  /tmp/pip-install-asbyc7hq/rhino3dm/src/bindings/../lib/opennurbs/opennurbs_lock.h:118:44: error: use of deleted function 'std::atomic<int>::atomic(const std::atomic<int>&)'
     std::atomic<int> m_lock_value = ON_Lock::UnlockedValue;
                                              ^
  In file included from /tmp/pip-install-asbyc7hq/rhino3dm/src/bindings/../lib/opennurbs/opennurbs_system.h:595:0,
                   from /tmp/pip-install-asbyc7hq/rhino3dm/src/bindings/../lib/opennurbs/opennurbs.h:29,
                   from /tmp/pip-install-asbyc7hq/rhino3dm/src/bindings/bindings.h:19,
                   from /tmp/pip-install-asbyc7hq/rhino3dm/src/bindings/bindings.cpp:1:
  /home/linuxbrew/.linuxbrew/Cellar/gcc/5.5.0_7/include/c++/5.5.0/atomic:616:7: note: declared here
         atomic(const atomic&) = delete;
         ^
  /home/linuxbrew/.linuxbrew/Cellar/gcc/5.5.0_7/include/c++/5.5.0/atomic:620:17: note:   after user-defined conversion: constexpr std::atomic<int>::atomic(std::atomic<int>::__integral_type)
         constexpr atomic(__integral_type __i) noexcept : __base_type(__i) { }
                   ^
  make[2]: *** [CMakeFiles/_rhino3dm.dir/build.make:95: CMakeFiles/_rhino3dm.dir/bindings/bindings.cpp.o] Error 1
  make[1]: *** [CMakeFiles/Makefile2:115: CMakeFiles/_rhino3dm.dir/all] Error 2
  make: *** [Makefile:103: all] Error 2
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-asbyc7hq/rhino3dm/setup.py", line 111, in <module>
      setup(
    File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 144, in setup
      return distutils.core.setup(**attrs)
    File "/usr/lib/python3.8/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/usr/lib/python3/dist-packages/wheel/bdist_wheel.py", line 223, in run
      self.run_command('build')
    File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/usr/lib/python3.8/distutils/command/build.py", line 135, in run
      self.run_command(cmd_name)
    File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/tmp/pip-install-asbyc7hq/rhino3dm/setup.py", line 45, in run
      self.build_extension(ext)
    File "/tmp/pip-install-asbyc7hq/rhino3dm/setup.py", line 99, in build_extension
      system("make")
    File "/tmp/pip-install-asbyc7hq/rhino3dm/setup.py", line 20, in system
      raise RuntimeError('The command "{}" exited with {}'.format(cmd, rc))
  RuntimeError: The command "make" exited with 2
  ----------------------------------------
  ERROR: Failed building wheel for rhino3dm
```

mcneel-build, 2021-03-24:
Linked with [RH3DM-141](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-141)

pearswj, 2021-03-24:
I just published a `manylinux2014` wheel for rhino3dm. Please try updating pip and installing again -- it should be a lot faster!

For the record, this build failed because gcc is too old.

sbaer, 2021-03-25:
@seghier it would be nice to know if this new wheel works for you

seghier, 2021-03-25:
I don't test this yet
Do youn mean i will use the same command
pip3 install rhino3dm ? Or the wheel have a different name?

sbaer, 2021-03-25:
pip3 install rhino3dm
Should work now

sbaer, 2021-04-03:
Closing issue. I’ll assume this was fixed unless I hear otherwise

closed 2021-04-03 by sbaer