mcneel/rhino3dm #358: Brep tp mesh?

seghier, 2021-03-26:
Hello
How to convert brep tp mesh?
In Ghpython there is : Rhino.Geometry.Mesh.CreateFromBrep but i don't find it in rhino3dm

mcneel-build, 2021-03-26:
Linked with [RH3DM-142](https://mcneel.myjetbrains.com/youtrack/issue/RH3DM-142)

sbaer, 2021-03-26:
rhino3dm contains a subset of RhinoCommon functionality that we give away for free in an open source toolkit. Creating meshes from Breps is not part of this set. You either need to use RhinoCommon running inside of Rhino or make an API call to a compute server which is really running RhinoCommon inside of Rhino.

closed 2021-03-26 by sbaer

seghier, 2021-03-26:
Thanks @sbaer 
I try this , **extrusion to mesh** work fine but **face to mesh** give none? why

```
import rhino3dm as rh
file = "......\\box.3dm"
rf = rh.File3dm.Read(file)
for obj in rf.Objects:
    geo = obj.Geometry
    if str(geo.ObjectType) == "ObjectType.Extrusion":
        print geo.GetMesh(rh.MeshType.Any)
        faces = geo.ToBrep(True).Faces
        for i in range(len(faces)):
            print faces[i]
            print faces[i].GetMesh(rh.MeshType.Any)
```

```
<rhino3dm._rhino3dm.Mesh object at 0x0000000002ADE5E0>
<rhino3dm._rhino3dm.BrepFace object at 0x0000000002ADE6C0>
None
<rhino3dm._rhino3dm.BrepFace object at 0x0000000002ADE6C0>
None
<rhino3dm._rhino3dm.BrepFace object at 0x0000000002ADE6C0>
None
<rhino3dm._rhino3dm.BrepFace object at 0x0000000002ADE6C0>
None

Process finished with exit code 0
```