mcneel/rhino3dm #365: No module named Point2d

KeithSloan, 2021-04-20:
Trying to use rhino3dm on a Mac get error

   

>        import rhino3dm as r3
>        File "/Applications/FreeCAD 0.19.1.app/Contents/Resources/lib/python3.8/site-packages/rhino3dm/__init__.py", line 6, in    <module>'
>        from` Point2d import *'
>        <class 'ModuleNotFoundError'>: No module named 'Point2d`

sbaer, 2021-04-20:
This looks like another case where something is messed up specifically with freecad and rhino3dm

KeithSloan, 2021-04-20:
Well "/Applications/FreeCAD 0.19.1.app/Contents/Resources/lib/python3.8/site-packages/rhino3dm/"

> Has

>  __init__.py			_rhino3dm.cpython-39-darwin.so
> __init__.pyi			py.typed
> __pycache__

And I assume rhino3dm is trying to import from Point2d.

FreeCAD keeps its Python libraries separate so I assumed I also have to install Point2d. 
So I tried pip3 install Point2d -t /Applications/FreeCAD_0.19.1.app/Contents/Resources/lib/python3.8/site-packages --upgrade --force-reinstall --no-cache-dir
Collecting Point2d
  Downloading point2d-0.0.1-py3-none-any.whl (17 kB)
Installing collected packages: Point2d
Successfully installed Point2d-0.0.1

But when I run I still get the error as originally reported.

KeithSloan, 2021-04-20:
I tried checking /Applications/FreeCAD_0.19.1.app/Contents/Resources/lib/python3.8/site-packages
and it does not have a Point2d just a point2d

sbaer, 2021-04-20:
It looks like the python 3.9 version of rhino3dm is installed when you really need the python 3.8 version of rhino3dm.

KeithSloan, 2021-04-21:
Where do I find the documentation that tells me what version of rhino3dm goes with which python version, I got the impression that any version of Python better than 3.7 was good to go.

If I run Python3 from the command line I get

> Python 3.8.2 (default, Dec 21 2020, 15:06:04) 
> [Clang 12.0.0 (clang-1200.0.32.29)] on darwin
> Type "help", "copyright", "credits" or "license" for more information`

My version of FreeCAD is built with 
Python version: 3.8.8

KeithSloan, 2021-04-21:
Upgrading Python 3.8.2 to match the 3.8.8 used by FreeCAD and reinstalling rhino3dm fixed the issue.

closed 2021-04-21 by KeithSloan

sbaer, 2021-04-21:
rhino3dm is a C extension library for python. These types of libraries are tied to specific major.minor releases of python.