mcneel/rhino3dm #373: Move CI builds to GitHub Actions

pearswj, 2021-05-19:
Migrate from CircleCI to GitHub Actions to make it easier to collaborate on build configuration and maintenance.

There are a lot of builds, so we could use [path filters](https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#example-including-paths) to avoid running unnecessary builds (e.g. only build dotnet project on changes to `src/dotnet/**` or `src/librhino3dm_native/** and vice-versa for python and javascript builds.

We also have a suite of internal builds that are used to build all of the Python wheels for a release. This could also be migrated to GitHub Actions to be [triggered manually](https://docs.github.com/en/actions/managing-workflow-runs/manually-running-a-workflow).

fraguada, 2021-05-31:
- [ ] Migrate from CircleCI to GitHub Actions
- [ ] use path filters to avoid running unnecessary builds
- [ ] build all of the Python wheels for a release, automate releases from manual trigger

fraguada, 2021-06-03:
Note: ci build won't go through builds at the moment if manually run from the ui. This will be fixed by checking the kind of event had activated the workflow.

closed 2021-07-16 by fraguada