mcneel/rhino3dm #376: docgen doesn't create correct ts definition for rhino3dm.js

fraguada, 2021-05-20:
Example: 

https://github.com/mcneel/rhino3dm/blob/master/src/bindings/bnd_surface.h#L33
```javascript
BND_TUPLE FrameAt(double u, double v);
```

results in a definition coming from the RhinoCommon method.

pearswj, 2021-05-20:
If you investigate this, can you take a look at #349? It should make working with docgen a little easier.

fraguada, 2021-05-20:
Yes! That will be really useful.

fraguada, 2021-05-27:
Editing this issue because the same is the case for python. For example:
RhinoCommon: https://developer.rhino3d.com/api/RhinoCommon/html/M_Rhino_Geometry_Intersect_Intersection_LinePlane.htm

```csharp
public static bool LinePlane(
	Line line,
	Plane plane,
	out double lineParameter
)
```

Binding code:
```cpp
BND_TUPLE BND_Intersection::PlanePlane(const BND_Plane& planeA, const BND_Plane& planeB)
{
  ON_Line line;
  bool success = ON_Intersect(planeA.ToOnPlane(), planeB.ToOnPlane(), line);
  BND_TUPLE rc = CreateTuple(2);
  SetTuple(rc, 0, success);
  SetTuple(rc, 1, line);
  return rc;
}
```

rhino3dm.py: https://mcneel.github.io/rhino3dm/python/api/Intersection.html#rhino3dm.Intersection.LinePlane
[image]

rhino3dm.js: https://mcneel.github.io/rhino3dm/javascript/api/Intersection.html#.linePlane
[image]

sbaer, 2021-05-27:
we need to handle functions with `out` parameters as returning tuples

PR #391 (2021-05-27): Update docgen to work with out parameters
fixes #376
closed 2021-05-27 by sbaer