mcneel/rhino3dm #405: Code-signed binaries for Python version

jesterKing, 2021-07-07:
On the mac after installing rhino3dm one can't directly load the rhino3dm module when it isn't code-signed.

Currently I need to instruct my users of my Blender add-on to manually remove the quarantine flag, but it would be better if the native binary part is code-signed.

I have such instructions on my release page

https://github.com/jesterKing/import_3dm/releases/tag/0.0.8

dbelcher, 2021-08-19:
Added this PR: https://github.com/mcneel/rhino3dm/pull/415, which I would like help testing. Currently, this is only set up to code-sign the python library: I'd like to test that first.

Once we have that working, I'd like to turn this run step into a reusable action, which I believe is called a "composite run step" (https://docs.github.com/en/actions/creating-actions/creating-a-composite-run-steps-action)
that could be called from any of the platform-specific build step. I would also be fine with putting the contents of the script/codesign-native.sh into this reusable step (rather than in a file itself), but I was hesitant to do that until we'd figured this out. I think this is just my lack of experience with GitHub actions...but let's try it together.

dan-rigdon-bel, 2021-11-15:
@fraguada Is this working now?

fraguada, 2021-11-24:
How would I check this? The workflow release was run successfully on all the py packages: https://github.com/mcneel/rhino3dm/actions/runs/1476178680

fraguada, 2023-10-05:
This should already be fixed.

closed 2023-10-05 by fraguada