mcneel/rhino3dm #419: Wrong Brep normal

devngc, 2021-10-01:
Hi team,

I probably have found in a bug in how Normal is returned in rhino3dm. When I am accessing the normal of the face shown in the image below. It is returned as 0.0, 0.0, 1.0. I would expect 0.0, 0.0, -1.0 in this case.
[image]
Following are the rhino file and the script I used.

```
import rhino3dm

path = r"C:\Users\devan\Desktop\test.3dm"
file_3dm = rhino3dm.File3dm.Read(path)

objects = [obj for obj in file_3dm.Objects]
print(
    f'Normal at center is {objects[0].Geometry.Faces[0].UnderlyingSurface().NormalAt(0.5,0.5)}')
```
[test.zip](https://github.com/mcneel/rhino3dm/files/7267033/test.zip)

sbaer, 2021-10-01:
Brep faces have a reversed flag that needs to be exposed in python and javascript. We call this property OrientationIsReversed in RhinoCommon

https://developer.rhino3d.com/api/RhinoCommon/html/P_Rhino_Geometry_BrepFace_OrientationIsReversed.htm

devngc, 2021-10-01:
Thanks for your response @sbaer,
Am I correct to understand that you're going to expose that?

sbaer, 2021-10-01:
yes, this property needs to be exposed

fraguada, 2023-10-05:
Just added: https://github.com/mcneel/rhino3dm/commit/5043ce81ab812dbd6ae6ad72a1d7c86a02f61b2a

fraguada, 2023-10-10:
This has now been added to py and js in 8.0.0-beta3

closed 2023-10-10 by fraguada