mcneel/rhino3dm #431: Mesh.createFromThreejsJSON type and arguments

nikokozak, 2021-11-17:
I'm trying to replicate one of the examples, namely the OBJ loader. The following line makes it seem like meshes can be created from THREEJS geo:

`const rhinoMesh = rhino.Mesh.createFromThreejsJSON( { data: mesh.geometry } )`

Nonetheless, the documentation states that `Mesh.createFromThreejsJSON` takes no arguments, and when used with TypeScript, it throws an error (no arguments expected). 

What would be the best way to go about working/converting THREEJS geometries + r3dm meshes?

fraguada, 2021-11-19:
This is the correct way to do it, even though our docs don't seem to show the correct arguments. We will take a look to see how to fix the documentation issue.

fraguada, 2023-02-07:
We are not catching cases where there is a specific method for js that does not exists for dotnet and python. I'm looking into fixing this.

fraguada, 2023-03-13:
Just published rhino3dm.js 8.0.0 where I manually fixed the definitions for these special cases where there is a particular method (related to threejs) in js that isn't in other libraries. https://www.npmjs.com/package/rhino3dm

closed 2023-03-13 by fraguada