mcneel/rhino3dm #433: rhino3dm.js Module type definition aka rhino3dm.d.ts is a really mess!

KaivnD, 2021-11-22:
As you can see in the title, so much error in type definition file `rhino3dm.d.ts`.
Few yeas ago, I found this module, which really got my attention. I like it, but when I use it with typescript, oh no, so much to `// @ts-ignore`, it really got me mad, so I decided try to make things right, only if you guys let me, I can make pr for this.

sbaer, 2021-11-22:
Feel free to submit a PR. This would require updating the ts file generator. The code that creates the ts file is at
https://github.com/mcneel/rhino3dm/blob/main/src/docgen/JavascriptClass.cs#L206

fraguada, 2021-12-01:
related to #431

fraguada, 2023-02-07:
Closing this as we're tracking the fixes to d.ts in #470 and #431

closed 2023-02-07 by fraguada