mcneel/rhino3dm #434: Python wheels are gzipped instead of zipped

jesterKing, 2021-11-29:
For package upload to work properly with `python -m twine upload  dist/*` the wheels should be zipped instead of gzipped. The gzip format causes the twine upload to fail with BadZipFile exception.

Additionally the wheel files until 7.7.0 on https://pypi.org/project/rhino3dm have the rhino3dm and rhino3dm-x.yy.z-dist-info folders at the top. The current release workflow has an extra level in there that doesn't help. It appears that this extra level is itself an archive. This level appears to be the wanted zip file.

[image]

[image]

fraguada, 2021-12-01:
These are gzipped with the artifacts action used by upload-artifact: https://github.com/actions/toolkit/blob/master/packages/artifact/docs/additional-information.md#is-my-artifact-compressed

jesterKing, 2022-03-22:
After extracting the all wheel zip on the Mac I used the following zsh script to rezip all the wheels

``` zsh
for i in *.whl; do                          
    fldr=`basename -s .whl $i`
    mkdir -p $fldr
    tar xvzf $i -C $fldr
    mv $i $i.bak
    zip -r $i $fldr/*
done
```

fraguada, 2025-02-28:
We handle this manually for now in the publishing steps.

closed 2025-02-28 by fraguada