mcneel/rhino3dm #435: Example request. BREP boolean example for JS/wasm

mmiscool, 2021-12-25:
Hello,
I was hoping that there might be an example for creating 2 primitive objects like a cube and a sphere, performing a boolean operation using the two objects and visualizing the output.

jesterKing, 2021-12-25:
The `rhino3dm` library has no geometry operation functionality. This library is primarily intended for _reading_ and _writing_ the Rhinoceros 3D file format.

If you need the Rhino geometry kernel functionality you should consider setting up Rhino.Compute to use in conjunction with your service.

closed 2021-12-25 by jesterKing

sbaer, 2021-12-27:
We do have samples for performing this type of operation in the browser using Rhino.Compute as a backend service
https://github.com/mcneel/rhino-developer-samples/tree/7/compute/js/SampleBrepBoolean