mcneel/rhino3dm #436: define a new instance definition

Gaspard-Bourgeois, 2022-01-08:
I don't succeed in creating new instance definition by code.

Can you update scripts to match with the documentation of Add method bellow :
https://mcneel.github.io/rhino3dm/python/api/File3dmInstanceDefinitionTable.html

Currently only idef can be added and I don't find any way to create a new idef from objects.

Zhanchenhaha, 2022-05-21:
hello,Is your problem solved? I have the same problem in C#

Gaspard-Bourgeois, 2022-05-21:
I didn't find any solution, except starting from a template

fraguada, 2023-12-19:
I've started work to this here: https://github.com/mcneel/rhino3dm/tree/luis/addInstanceDefinition

fraguada, 2024-03-21:
This has been added and will be available in rhino3dm 8.6.0

fraguada, 2024-04-05:
@Gaspard-Bourgeois @Zhanchenhaha This has been fixed in rhino3dm. You can get it on 8.6.0-beta1 [[dotnet](https://www.nuget.org/packages/Rhino3dm/8.6.0-beta1)][[py](https://pypi.org/project/rhino3dm/8.6.0b1/)][[js](https://www.npmjs.com/package/rhino3dm/v/8.6.0-beta1)]