mcneel/rhino3dm #438: import rhino3dm fails

KeithSloan, 2022-01-13:
I have tried to install rhino3dm in a library where FreeCAD will find it with
````
pip3 install -t /Applications/FreeCAD_19.2.app/Contents/Resources/lib/python3.8/site-packages rhino3dm --upgrade --force
````
But import rhino3dm fails with
````
import rhino3dm as r3
  File "/Applications/FreeCAD_19.2.app/Contents/Resources/lib/python3.8/site-packages/rhino3dm/__init__.py", line 5, in <module>
    from ._rhino3dm import *
<class 'ModuleNotFoundError'>: No module named 'rhino3dm._rhino3dm'
````
__init__.py looks like
````
import sys
if sys.version_info.major==2:
    from _rhino3dm import *
else:
    from ._rhino3dm import *

__version__ = '7.7.0'

Point2d.__str__ = lambda self: "{},{}".format(self.X, self.Y)
Point3d.__str__ = lambda self: "{},{},{}".format(self.X, self.Y, self.Z)
Vector2d.__str__ = lambda self: "{},{}".format(self.X, self.Y)
Vector3d.__str__ = lambda self: "{},{},{}".format(self.X, self.Y, self.Z)
````
And the directory does not contain any such files
````
keithsloan@Keiths-iMac rhino3dm % ls
__init__.py			_rhino3dm.cpython-39-darwin.so
__init__.pyi			py.typed
__pycache__
````

jesterKing, 2022-01-13:
According your logs you are using Python 3.8, but the rhino3dm module you're loading is for Python 3.9 - that will not work, you _need_ to have the correct version installed.

KeithSloan, 2022-01-14:
Okay (Although it says experimental) I can see what versions are available to install with pip using
pip index versions rhino3dm
Is there some way/where I can check which versions go with which python?

KeithSloan, 2022-01-14:
I am trying to document the install requirements for my FreeCAD workbench that uses rhino3dm. FreeCAD runs on Windows, Linux, MacOS and different versions of FreeCAD use different versions of Python. It is not very satisfactory to specific links to wheels.

I just tried installing 7.6.0 and then 0.16.0 and still have the same problem.

jesterKing, 2022-01-14:
I am not sure why you are trying older builds instead of the new that I linked. 7.11.0 is current most up-to-date.

To install the rhino3dm for the correct version you'll have to use the same Python version as used if FreeCAD, in your case on MacOS that would be Python 3.8. I believe at the moment Python 3 would default to Python 3.9, so you'll have to ensure you use `pip` that goes with the correct Python version 3.8.

KeithSloan, 2022-01-14:
I was trying older versions as I thought they might have been built with older python.
I tried

Okay I tried
````
python3.8 -m ensurepip --default-pip
pip install -t /Applications/FreeCAD_19.2.app/Contents/Resources/lib/python3.8/site-packages rhino3dm --upgrade --force
Collecting rhino3dm
  Using cached rhino3dm-7.7.0-cp39-cp39-macosx_10_15_x86_64.whl (3.7 MB)
Installing collected packages: rhino3dm
````
Still does not work.

KeithSloan, 2022-01-14:
Okay I got it to work with
````

jesterKing, 2022-01-15:
So this now works for you in FreeCAD on MacOS as well?

KeithSloan, 2022-01-15:
Yes using the last approach.

Are the warnings okay to ignore?

jesterKing, 2022-01-17:
They look like warnings from PIP, not from our code. You could report it to the link mentioned in the log.

closed 2022-01-17 by jesterKing

KeithSloan, 2024-12-07:
What version of rhino3dm would work with python 3.11.10?

Thanks

KeithSloan, 2024-12-09:
Okay

As FreeCAD 1.0.0 is using python3.11.10

I managed to fix by 1 installing python3.11 on Mac using
```
homebrew python@3.11
```
Then installed rhino3dm
```
pip install rhino3dm
```
Then in FreeCAD path with the --no-cache option
```
python3.11 -m pip install rhino3dm --upgrade -t'/Applications/FreeCAD 1.0.0.app/Contents/Resources/lib/python3.11/site-packages' --no-cache
Collecting rhino3dm
  Downloading rhino3dm-8.9.0-cp311-cp311-macosx_14_0_universal2.whl.metadata (1.3 kB)
Downloading rhino3dm-8.9.0-cp311-cp311-macosx_14_0_universal2.whl (8.4 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.4/8.4 MB 4.3 MB/s eta 0:00:00
Installing collected packages: rhino3dm
Successfully installed rhino3dm-8.9.0
```