mcneel/rhino3dm #440: Apple Silicon: Support x86_64 and arm64 Universal binary

dan-rigdon-bel, 2022-01-21:


jesterKing, 2022-03-22:
I didn't do a universal build yet, I can test that next, but arm compile support is now in with #462

mit200, 2022-05-12:
hi @jesterKing we would like to use the library with a raspberry pi model 4 b. Is there some known way to compile it to arm64?
Thank you in advance!

jesterKing, 2022-05-13:
@mit200 I have to break out my rpi4 and fiddle around a bit, but in the meantime have a look at https://github.com/mcneel/opennurbs/pull/13 and make the changes locally.

mit200, 2022-05-13:
@jesterKing thanks for the input! Sadly, I don't have the environment since I'm using it for a .NET project. Tried some stuff out but wasn't really able to get it going (I'm getting a cmake compile error on my windows system since arm is configured). Is there any way you could provide a prerelease with compiling it for ARM? Happy to test it out :)
[image]

jesterKing, 2022-05-13:
I don't know how to cross-compile this on Windows for rpi4. My rpi4 hardware is currently still somewhere in boxes, it'll be some time before I get to it.

You could try finding ways to emulate an arm64 system so that you could install the raspberry os and compile via that.

mit200, 2022-05-13:
@jesterKing sounds good thanks! In the meantime we'll try it on our rpi4 and maybe emulate an arm64 with docker and provide feedback.

fraguada, 2023-03-24:
With the 8.0.0 release we've added universal binaries for macOS .net builds. At this time there are no arm64 linux runners available on GitHub. I'll create a new issue for this.