mcneel/rhino3dm #454: 7.14 won't install through pip under Python 3.8

fstwn, 2022-03-01:
Hey, just for reasons of completeness I'm opening this issue, referencing [this discussion](https://github.com/mcneel/compute.rhino3d/pull/457#issuecomment-1055593633) and also tag @sbaer .

Unfortunately, when trying to update or install using pip, an error is thrown:
```
(ddu_ias_research) PS C:\Users\EFESTWIN\Desktop> pip --version
pip 22.0.3 from C:\Users\EFESTWIN\Anaconda3\envs\ddu_ias_research\lib\site-packages\pip (python 3.8)
(ddu_ias_research) PS C:\Users\EFESTWIN\Desktop> pip install rhino3dm
Collecting rhino3dm
  Using cached rhino3dm-7.14.0-cp38-cp38-win_amd64.whl (2.0 MB)
ERROR: rhino3dm has an invalid wheel, rhino3dm has an invalid wheel, .dist-info directory not found
```
Haven't tried using other Python versions though.

sbaer, 2022-03-01:
I see that too. Working on a fix

sbaer, 2022-03-02:
uploaded 7.14.1 to pypi. This looks like it is working now

closed 2022-03-02 by sbaer